Ysurac/openmptcprouter #757: openmptcprouter VPS installation error

8252610, 2019-12-22:
## Expected Behavior

When running the wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh
Error occurred
/tmp/shadowsocks-libev-3.3.3.tar 100%[==========================================================>] 1.84M 13.9KB/s in 2m 45s

2019-12-22 11:36:05 (11.4 KB/s) - '/tmp/shadowsocks-libev-3.3.3.tar.gz' saved [1930597/1930597]

--2019-12-22 11:36:05-- https://raw.githubusercontent.com/Ysurac/openmptcprouter-feeds/master/shadowsocks-libev/patches/020-NOCRYPTO.patch
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.228.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.228.133|:443... failed: Connection refused.

My server is alicloud China

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2019-12-22:
Run it again.
Having a mirror for Asia is in todo list.

8252610, 2019-12-22:
thank you
Every time I run, I pray to my God Monkey King
But I've run it 10 times

8252610, 2019-12-22:
[image]
当我运行乌班图脚本
My operating system Ubuntu 18.04

Ysurac, 2019-12-22:
I don't see any error here.

8252610, 2019-12-22:
It's been on this screen for an hour
I'm installing Python 3.7 for the system and running the script

8252610, 2019-12-22:
[image]
Error after waiting

8252610, 2019-12-22:
[image]
I am ready to replace the local operating system test

Ysurac, 2019-12-22:
Compiler is killed, check using dmesg why. Maybe a VPS limit.

suyuan168, 2019-12-22:
hi, 
First, you have to solve your network restrictions. Networks in China do not support shadowsocks. Then you have to check the system for environmental problems. Follow the installation tutorial. Then check your system log and resolve it.

8252610, 2019-12-23:
YES, 
I can't speak English. The baidu translation I use may have grammar problems
My purchase of alicloud's French VPS solves the problem of shadowlocks
Now we are solving the environment problem of Alibaba cloud Ubuntu system
My server in Amazon can complete VPS installation
thank you

8252610, 2019-12-25:
I successfully installed VPS
Purchase VPS system is Debian 9 (operators do not have Debian 10) and upgrade to Debian 10
Run debian10 VPS installation script
Thank you, author. You're so good

diogosilvan, 2020-01-23:
Exact same error here. Debian 10 and Ubuntu 18.04 LTS same error.
Problem present in Vultr and Locaweb (in Brazil)

Ysurac, 2020-01-23:
What is the error ? A "connection refused" to some Github hosted content ?

diogosilvan, 2020-01-23:
> What is the error ? A "connection refused" to some Github hosted content ?

 Failed building wheel for uvloop
  Running setup.py clean for uvloop
...
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-t09g_o3_/uvloop/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-ssgnw5sy/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-t09g_o3_/uvloop/

Ysurac, 2020-01-23:
With a killed compiler before that ? Check with ```dmesg``` why it's killed.

diogosilvan, 2020-01-23:
> With a killed compiler before that ? Check with `dmesg` why it's killed.

dmesg does not provide any information about the compiler.
The problem is not directly in the MPTCP installation script.
In a newly installed VPS (installed only the tools needed for python) I try to install "pip3 install uvloop" I have the same error.

diogosilvan, 2020-01-23:
Solved, the problem was low memory that was killing the compiler.
On a 512MB RAM machine it doesn't work anymore, only on 1GB above.

taojiji, 2020-02-01:
我也是遇到同一个问题，阿里云主机

taojiji, 2020-02-01:
I have the same problem, aliyun host

closed 2020-03-10 by Ysurac