Ysurac/openmptcprouter #760: Error : daemon.err omr-tracker-ss

flimo44, 2019-12-22:
I have this error in systeme Log

daemon.err omr-tracker-ss[20051]: Failed to establish connection

## Steps to Reproduce the Problem

  1. Many time in system log
  2. I have also in UpenMptcprouter : no ipv6 accès
  3. Now ping IP V6 on my VPS it's OK after change Ip6tables 

Specifications
OpenMPTCProuter version: 0.53
OpenMPTCProuter VPS version:0.1005
OpenMPTCProuter platform: x86_64

Ysurac, 2019-12-22:
It's because check of Shadowsocks-libev status doesn't success.

flimo44, 2019-12-22:
I have also juste before the message :
daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)

I don't know where to look. Is this due to the problem of IP V6 accès?

Ysurac, 2019-12-22:
Possible, if IPv6 is enabled there is a test to get public IPv6.

flimo44, 2019-12-22:
Yes IPV6 is enabled. I have install  apt install ipv6table was missing in script  on my VPS. Re-run the script.
script installation on my VPS was O and have now the 0.1005 version. Just lost my ipv6 accès. I resolved it by import a new default ip6tables. Now if i do a ping gateway or google V6 ip work fine.

Ysurac, 2019-12-22:
Firewall is done by shorewall. Shorewall6 is used for IPv6..
What is you VPS OS ?

flimo44, 2019-12-22:
My VPS OS is debian 10. (OVH)
I test a traceroute -6 google.fr  on my router openmptcp . The traceroute  stop at the IP V6 VPS reponse  ok . Nothing after.

Ysurac, 2019-12-22:
If you have custom ip6tables rules this is a problem for IPv6.

flimo44, 2019-12-22:
Yes , but if i use the previous iptables it was impossible to ping ip6 public on my VPS :(.
Have you a ip6tables i can use and test ?

Ysurac, 2019-12-22:
As I said, shorewall6 must be used.

flimo44, 2019-12-22:
OK. shorewall6 is installed on my VPS. I try to understand how use it correctely and where is the problem. It's not easy because new for me. The shorewall check is OK. I continu my inverstigation.
Merci Ysurac et bonne soirée

flimo44, 2020-01-01:
Hello,

I finaly solved issue by correcting in / etc/shorewall6/rules ( VPS )

Just Adding this 2 lines:

ACCEPT net $FW ipv6-icmp
ACCEPT $FW vpn ipv6-icmp

and do : shorewall6 restart

Now i can Ping V6 adress from VPS
curl -6 www.google.com : OK

And stat openmptcprouter is now OK

Ping and traceroute OK also from router.

closed 2020-01-01 by flimo44