Ysurac/openmptcprouter #761: RPi 4 settings no persist between reboots

Natan4ik, 2019-12-23:
## Expected Behavior

Settings persist between reboots.

## Actual Behavior

Settings no longer persist between reboots (seems like everything resets to factory defaults on every reboot)

I installed fresh build 0.53 on RPi 4.
After making changes and subsequent rebooting, all previously made changes are erased.

SD card is new, does not have any problems.
The image is installed on the card using latest baleaEtcher.

## Steps to Reproduce the Problem

  1. Install openmptcprouter v0.5x squashfs-factory on RPi 4
  2. Set root password
  3. Reboot
  4. Attempt to connect via ssh
  5. Receive a message about the need to change the password:
```
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
```
  6. Result of "df -h"
```
root@OpenMPTCProuter:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                46.0M     46.0M         0 100% /rom
devtmpfs                  1.7G         0      1.7G   0% /rom/dev
tmpfs                     1.9G    368.0K      1.9G   0% /tmp
tmpfs                     1.9G    184.0K      1.9G   0% /tmp/root
overlayfs:/tmp/root       1.9G    184.0K      1.9G   0% /
/dev/mmcblk0p1           63.9M     16.5M     47.4M  26% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
```

  7. dmesg log
```
[    9.228911] mount_root: failed to mount -t f2fs /dev/loop0 /tmp/overlay: No such device
[    9.239681] mount_root: overlay filesystem has not been fully initialized yet
[    9.249627] mount_root: switching to f2fs overlay
[    9.256924] mount_root: switching to f2fs failed - fallback to ramoverlay

```

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.53 r011546-a74095c68c-brcm2708-bcm2711
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter platform: RPi4 64-bit

user747, 2019-12-25:
try using ext4 image instead.
squashfs has issues sometimes

Natan4ik, 2019-12-26:
ext4 works correctly, but we have frequent cases of a sharp power failure (this is not possible to avoid).
The next time the device is turned on, ext4 is mounted in ro.

Ysurac, 2019-12-26:
It's a problem with RPI that is very sensitive to power failure and I've not real solutions for that yet. You should use an UPS.

user747, 2019-12-30:
> we have frequent cases of a sharp power failure
i agree it would be good to invest in a UPS solution

although i have my rpi connected to a pure sine wave UPS, using ext4, and have encountered some times where the file system went read only. tested sd card to be sure using badblocks and it passed, so my guess is a misconfiguration issue on my end. openwrt upstream rpi firmware is also a few months behind.

Ysurac, 2019-12-30:
There is no changes about SD card in firmware updates.
I had read only problems on RPI only with some cheap SD card, but FS going to read only is a know RPI issue.

closed 2020-05-18 by Ysurac