Ysurac/openmptcprouter #762: Address not available

bosseb, 2019-12-24:
Hostname	OpenMPTCProuter
Model	Beelink BT3 PRO
Architecture	Intel(R) Atom(TM) x5-Z8350 CPU @ 1.44GHz
Firmware Version	openmptcprouter v0.53 r0+11546-a74095c68c / LuCI Master git-19.323.39847-15816e6
VPS 0.1005 SCALEWAY

Hello , i have these errors : (what's the problem ?)
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=487, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=483, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=469, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=455, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=475, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=479, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=407, dest=[ff0e::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=487, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=483, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=469, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=455, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=475, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=479, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=407, dest=[ff05::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=487, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=483, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=469, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=455, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=475, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=416, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=479, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=407, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:11:07 2019 daemon.err miniupnpd[5141]: try_sendto failed to send 39 packets
Tue Dec 24 08:11:18 2019 daemon.warn miniupnpd[5141]: SSDP packet sender [::ffff:169.254.25.139]:45652 (if_index=-1) not from a LAN, ignoring
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
Tue Dec 24 08:11:37 2019 daemon.err miniupnpd[5141]: sendto(udp_notify=8, [fd50:92ee:5c26::1]): Address not available
[... 448 lines cut ...]
Tue Dec 24 08:14:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=471, dest=[ff02::c]:1900): sendto: Address not available
[... cut ...]
Tue Dec 24 08:14:07 2019 daemon.err miniupnpd[5141]: try_sendto(sock=8, len=407, dest=[ff02::c]:1900): sendto: Address not available
Tue Dec 24 08:14:07 2019 daemon.err miniupnpd[5141]: try_sendto failed to send 39 packets
Tue Dec 24 08:14:24 2019 daemon.warn miniupnpd[5141]: SSDP packet sender [::ffff:169.254.25.139]:45652 (if_index=-1) not from a LAN, ignoring

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2019-12-24:
Seems to be some local IPv6 addresses not available.
No idea why, I don't use miniupnpd and this should really not be used.

bosseb, 2019-12-24:
i don't know what 's miniupnpd is it ... so i need to disable it ? how can i disable it ?

Ysurac, 2019-12-24:
It's UPnP in Network->UPnP. Disabled by default.

bosseb, 2019-12-24:
thank you i want to know if i need to enable 
Activer la fonctionnalité NAT-PMP

Activer le mode sécurisé
??

closed 2020-05-18 by Ysurac