Ysurac/openmptcprouter #763: IPv6 Routing

bjoerrrrn, 2019-12-24:
## Expected Behavior

IPv6 /64 Subnet on VPS ist available. When i turn on IPv6 Support i want IPv6 Connectivity. 

## Actual Behavior

Nothing happens. Even if i assign a he.net Tunnel Subnet - i Made it Work that my Clients get a IPv6 from the Subnet supplied. Traceroute6 gets stuck on router IP.

## Steps to Reproduce the Problem

  1. Activate IPv6 in Gateway Settings
  2.
  3.

## Specifications

  - OpenMPTCProuter version: 0.53
  - OpenMPTCProuter VPS version: 0.005
  - OpenMPTCProuter platform: x86_64

Can you supply a Tutorial for IPv6 Routing please? :-(

Ysurac, 2019-12-25:
You only need to have working IPv6 on the VPS (never tested with a he tunnel on this side) and check Activate IPv6 box in the wizard.

bjoerrrrn, 2019-12-25:
VPS IPv6 is working fine. I can ping and tracereroute.

When i just enable IPv6 in the Wirzard i get:
[image]

I get no failures in the system log. Don`t know where i can start to investigate. :-(

Ysurac, 2019-12-25:
On the VPS it's native IPv6 or a tunnel ? If it's a tunnel Shorewall must be configured.
You can also check on the router with ```ip a``` that the tunnel use correct VPN settings.

flimo44, 2019-12-26:
hello,
i have the same  problem and error. ( VPS OVH dedian 10 )

i don't see what's wrong :/ :
[Vps_ip_a.txt](https://github.com/Ysurac/openmptcprouter/files/4003403/Vps_ip_a.txt)

Sorry i dont anderstand when you say : it's native IPv6 or a tunnel ?  "If it's a tunnel Shorewall" must be configured.
My VPS has a native IP V6.  I can ping external V6 IP. ( so, i don't need ShoreWall !). Also, Shorewall seems to be intalled fine on my VPS. 
( Désolé , mais là je  seiche et je ne vois pas ce que j'ai pu mal configurer. je pense mais sans certitude que mon problème est arrivé suite à la dernière mise à jour du VPS. Je ne pouvais plus pinguer les V6 externe ). J'ai donc backupé mon ip6tables et importé un autre qui me permet à nouveau de pinguer les V6 du VPS. Si je lance un ping V6 de mon routeur openmtpc ça bloque à l'ip V6 de mon VPS. Donc mon VPS ne route pas mais ip6 venant de mon routeur )

Ysurac, 2019-12-26:
HE give IPv6 over a tunnel, so not native.
ip addresses config seems good.
Do you have something about IPv6 on VPS using ```dmesg``` ?
Do you have also an error on router status page about IPv6 ?
Glorytun TCP is the used VPN ?

flimo44, 2019-12-26:
I just see on dmesg VPS :
[420847.979156] Shorewall:net-fw:DROP:IN=ens3 OUT= MAC=fa:16:3e:e5:b0:2d:2e:ba:35:52:84:5d:08:00 SRC=81.22.45.65 DST=51.75.249.37 LEN=68 TOS=0x00 PREC=0x00 TTL=51 ID=28911 PROTO=ICMP TYPE=3 CODE=10 [SRC=51.75.249.37 DST=81.22.45.65 LEN=40 TOS=0x00 PREC=0x00 TTL=48 ID=0 DF PROTO=TCP SPT=37090 DPT=54991 WINDOW=0 RES=0x00 ACK RST URGP=0 ]
[420869.334008] ip6tables[DOS]: IN=omr-6in4 OUT= MAC=2e:ba:35:52:84:5d:fa:16:3e:e5:b0:2d:08:00:45:c0:00:6d:9f:15 TUNNEL=10.255.255.2->10.255.255.1 SRC=fd4e:3636:916e:0000:0000:0000:0000:0001 DST=2001:41d0:0305:2100:0000:0000:0000:660c LEN=89 TC=0 HOPLIMIT=64 FLOWLBL=655287 PROTO=UDP SPT=8591 DPT=53 LEN=49

On routeur status i have : no ip V6 acces

Yes i use Glorytun  proto : tcp

Ysurac, 2019-12-27:
There is no reason to have some ip6tables logs, remove any iptables rules you added and restart shorewall and shorewall6.

flimo44, 2019-12-27:
I remove all ip6tables ( ip6tables -F ) restarted shorewall and shorewall6 . 
Always  : not ip6 acces 

[image]

Now only Shorewall message in dmesg :

[471151.862999] Shorewall:vpn-net:ACCEPT:IN=omr-6in4 OUT=ens3 MAC=00:00:00:00:66:0c:ff:02:00:00:00:00:00:00:45:c0:00:64:af:1b TUNNEL=10.255.255.2->10.255.255.1 SRC=fd4e:3636:916e:0000:0000:0000:0000:0001 DST=2001:41d0:0002:7fc0:0000:0000:0000:0004 LEN=80 TC=0 HOPLIMIT=63 FLOWLBL=961896 PROTO=TCP SPT=24971 DPT=80 WINDOW=24400 RES=0x00 SYN URGP=0
[471152.873931] Shorewall:net-fw:DROP:IN=ens3 OUT= MAC=fa:16:3e:e5:b0:2d:2e:ba:35:52:84:5d:86:dd SRC=2001:41d0:0305:2100:0000:0000:0000:0001 DST=2001:41d0:0305:2100:0000:0000:0000:660c LEN=72 TC=0 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=136 CODE=0
[471155.937451] Shorewall:net-fw:DROP:IN=ens3 OUT= MAC=fa:16:3e:e5:b0:2d:2e:ba:35:52:84:5d:86:dd SRC=fe80:0000:0000:0000:2cba:35ff:fe52:845d DST=2001:41d0:0305:2100:0000:0000:0000:660c LEN=72 TC=0 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0

Ysurac, 2019-12-27:
It's an update and not a fresh install on the VPS ?
Check ```/etc/shorewall6/params.net``` on the VPS, the interface with IPv6 address should be set.

I also see that an IPv6 route is received on the frebox interface, IPv6 route announce need to be disabled on any box/modem. Can you give me the result of ```ip -6 r``` on the box ? It's possible that there is a IPv6 conflict.

flimo44, 2019-12-27:
It's a VPS update
In /etc/shorewall6/params.net there is 1 line with : NET_IFACE=ens3
Yes i receive route announce with the 2 interfaces regulary. 

root@OpenMPTCProuter:~# ip -6 r
default from fe80::/64 dev 6in4-omr6in4 proto static metric 6 pref medium
default from fe80:a00:1::/48 dev 6in4-omr6in4 proto static metric 6 pref medium
fd4e:3636:916e::/64 dev eth0 proto static metric 2 pref medium
fe80::/64 dev 6in4-omr6in4 proto static metric 6 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev wan1 proto kernel metric 256 pref medium
fe80::/64 dev wan2 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80:a00:1::/64 dev eth0 proto static metric 2 pref medium
unreachable fe80:a00:1::/48 dev lo proto static metric 2147483647 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 6 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
root@OpenMPTCProuter:~#

Ysurac, 2019-12-27:
You have a problem with omr6in4 configuration on the router. local and remote address are inverted.

flimo44, 2019-12-27:
this is my omr6in4 conf :
[image]

I just  remove IPv6 routed prefix was before : fe80::a00:1/48 ( no change ). Try olso to invert it. 

I don't anderstand what's is wrong and how do :(

my ip -6 r now 
default from fe80::/64 dev 6in4-omr6in4 proto static metric 6 pref medium
fd4e:3636:916e::/64 dev eth0 proto static metric 2 pref medium
fe80::/64 dev 6in4-omr6in4 proto static metric 6 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev wan1 proto kernel metric 256 pref medium
fe80::/64 dev wan2 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 6 pref medium
root@OpenMPTCProuter:~#

Ysurac, 2019-12-27:
So it's working now.

flimo44, 2019-12-27:
no :(

Ysurac, 2019-12-27:
You can traceroute from the router ? I think curl -6 www.google.com work also. In this case it's working from the router. You can still have issues with Free IPv6 route announce.

flimo44, 2019-12-27:
Since i have clear ip6tables i can't ping ipv6 from my VPS.
I think the problem is on VPS . 
if i do a traceroute -6 from router i have :
traceroute to google.fr (2a00:1450:4007:815::2003), 30 hops max, 72 byte packets
1 2001:41d0:305:2xx0::660c (2001:41d0:305:xxxx::660c) 19.418 ms 19.276 ms 20.076 ms
2 2001:41d0:305:2xx0::660c (2001:41d0:305:xxxx::660c) 356.343 ms !H 3081.859 ms !H 3107.868 ms !H

2001:41d0:305:2xx0::660c is my vps ip6 adress

flimo44, 2019-12-31:
Hello,

I finaly solved issu by correcting in / etc/shorewall6/rules  ( VPS )

Add this 2 lines: 

ACCEPT      net         $FW        ipv6-icmp                                                                                                                                                                                                                         
ACCEPT      $FW         vpn        ipv6-icmp

dnwk, 2019-12-31:
My server has native IPv6 connectivity. However, once the debian buster script is installed, it lose IPv6 rightaway.

Ysurac, 2020-01-01:
> Just Adding this 2 lines:
> 
> ACCEPT net $FW ipv6-icmp
> ACCEPT $FW vpn ipv6-icmp
> 
@flimo44 This lines are already in shorewall6/rules, you updated from an old release ?

@dnwk check that your IPv6 is set on the interface set in /etc/shorewall6/params.net

flimo44, 2020-01-01:
Yes possible have (re)run old Scritp before anderstand passed in debian10.
Can you add a test in old scritp to controle debian version and stop it if debian < 10 with a notice to run new script ?  ( This will prevent others from making mistakes)

bjoerrrrn, 2020-01-01:
I`m still not able to use ipv6.

[image]

[image]

ping6 from and to the vps and from and to the router works.
ping6/traceroute6 on the vps to the web works.

Kernel Logs on the router is full of:
[ 2398.541280] Dead loop on virtual device 6in4-omr6in4, fix it urgently!
[ 2403.552194] Dead loop on virtual device 6in4-omr6in4, fix it urgently!
[ 2408.811317] Dead loop on virtual device 6in4-omr6in4, fix it urgently!
[ 2413.824062] Dead loop on virtual device 6in4-omr6in4, fix it urgently!
[ 2418.832955] Dead loop on virtual device 6in4-omr6in4, fix it urgently!

ip -6 a on router:
[image]

ip -6 route on router:
[image]

ip -6 a on vps:
[image]

ip -6 route on vps:
[image]

I don't get it :-(

Ysurac, 2020-01-01:
What is not working from the router ?
You checked interface in /etc/shorewall6/params.net ?

bjoerrrrn, 2020-01-01:
> What is not working from the router ?

ping6 to ipv6 external adresses and routing to the ipv6 internet...

> You checked interface in /etc/shorewall6/params.net ?

root@gateway:~# cat /etc/shorewall6/params.net
NET_IFACE=eth0

Ysurac, 2020-01-01:
shorewall6 is running and working correctly ? You have some rules if you do a ```ip6tables-save --list``` ?

bjoerrrrn, 2020-01-02:
root@gateway:~# ip6tables-save --list
ip6tables-save: unrecognized option '--list'
Look at manual page `ip6tables-save.8' for more information.
root@gateway:~# ip6tables --list
Chain INPUT (policy DROP)
target     prot opt source               destination
net-fw     all      anywhere             anywhere
vpn-fw     all      anywhere             anywhere
ACCEPT     all      anywhere             anywhere
Broadcast  all      anywhere             anywhere
Multicast  all      anywhere             anywhere
LOG        all      anywhere             anywhere             LOG level info prefix "Shorewall:INPUT:REJECT:"
reject     all      anywhere             anywhere            [goto]

Chain FORWARD (policy DROP)
target     prot opt source               destination
net_frwd   all      anywhere             anywhere
vpn_frwd   all      anywhere             anywhere
Broadcast  all      anywhere             anywhere
Multicast  all      anywhere             anywhere
LOG        all      anywhere             anywhere             LOG level info prefix "Shorewall:FORWARD:REJECT:"
reject     all      anywhere             anywhere            [goto]

Chain OUTPUT (policy DROP)
target     prot opt source               destination
fw-net     all      anywhere             anywhere
fw-vpn     all      anywhere             anywhere
ACCEPT     all      anywhere             anywhere
ACCEPT     all      anywhere             anywhere

Chain Broadcast (4 references)
target     prot opt source               destination
DROP       all      anywhere             external
DROP       all      anywhere             external
Chain Multicast (4 references)
target     prot opt source               destination
DROP       all      anywhere             ff00::/8

Chain dynamic (4 references)
target     prot opt source               destination

Chain fw-net (1 references)
target     prot opt source               destination
ACCEPT     udp      anywhere             anywhere             udp dpts:dhcpv6-client:dhcpv6-server
ACCEPT     all      anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     udp      anywhere             anywhere             udp dpt:domain /* DNS */
ACCEPT     tcp      anywhere             anywhere             tcp dpt:domain /* DNS */
ACCEPT     ipv6-icmp    anywhere             anywhere
ACCEPT     all      anywhere             anywhere

Chain fw-vpn (1 references)
target     prot opt source               destination
ACCEPT     all      anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     ipv6-icmp    anywhere             anywhere
ACCEPT     udp      anywhere             anywhere             udp dpt:domain
ACCEPT     all      anywhere             anywhere

Chain logdrop (0 references)
target     prot opt source               destination
DROP       all      anywhere             anywhere

Chain logflags (7 references)
target     prot opt source               destination
LOG        all      anywhere             anywhere             LOG level info ip-options prefix "Shorewall:logflags:DROP:"
DROP       all      anywhere             anywhere

Chain logreject (0 references)
target     prot opt source               destination
reject     all      anywhere             anywhere

[... 86 lines cut ...]
tcpflags   tcp      anywhere             anywhere
vpn-net    all      anywhere             anywhere

-------------------

i removed my ipv6 adresses from above.

root@gateway:~# /etc/init.d/shorewall6 status
● shorewall6.service - Shorewall IPv6 firewall
   Loaded: loaded (/lib/systemd/system/shorewall6.service; enabled; vendor preset: enabled)
   Active: active (exited) since Wed 2020-01-01 13:05:41 CET; 20h ago
 Main PID: 960 (code=exited, status=0/SUCCESS)
   Memory: 0B
   CGroup: /system.slice/shorewall6.service

Jan 01 13:05:41 gateway shorewall[960]: Shorewall configuration compiled to /var/lib/shorewall6/.start
Jan 01 13:05:41 gateway shorewall[960]: Starting Shorewall6....
Jan 01 13:05:41 gateway shorewall[960]: Initializing...
Jan 01 13:05:41 gateway shorewall[960]: Preparing ip6tables-restore input...
Jan 01 13:05:41 gateway shorewall[960]: Running /sbin/ip6tables-restore --wait 60...
Jan 01 13:05:41 gateway shorewall[960]: IPv6 Forwarding Enabled
Jan 01 13:05:41 gateway shorewall[960]: Setting up IPv6 Interface Forwarding...
Jan 01 13:05:41 gateway root[1187]: Shorewall6 started
Jan 01 13:05:41 gateway shorewall[960]: done.
Jan 01 13:05:41 gateway systemd[1]: Started Shorewall IPv6 firewall.

dnwk, 2020-01-03:
>  /etc/shorewall6/params.net

My IPv6 is set on that file. It shows eth0. In /etc/network I set it on eth0:0.  Does it matter? Also, on my VPS, I can't access any IPv6 site after script installation.

flimo44, 2020-01-03:
if you do :
/etc/init.d/shorewall6 stop 
=> can you ping public V6  from VPS ?
and :
curl -6 ifconfig.co 
=> give you  vps IP ?

/etc/init.d/shorewall6 start

dnwk, 2020-01-05:
> /etc/init.d/shorewall6 stop

This command itself doesn't restore IPv6 to my VPS

Ysurac, 2020-01-05:
@bjoerrrrn you can try to remove ```fe80::/10``` in /etc/shorewall6/snat and restart it ```/etc/init.d/shorewall6 restart```.
I need to check if I have a VPS with a fe80... gateway.

dnwk, 2020-01-06:
> 
> 
> if you do :
> /etc/init.d/shorewall6 stop
> => can you ping public V6 from VPS ?
> and :
> curl -6 ifconfig.co
> => give you vps IP ?
> 
> /etc/init.d/shorewall6 start

I tweak server IPv6 settings and seems to get server IPv6 back. But not sure why. I moved ipv6 settings from eth0:1 to eth0:0

lolo31660, 2020-04-16:
I've got the same issue on my VPS. Config cas OK, then after finishing my installation & tuning, I can't ping ipv6 adresses

Ysurac, 2020-04-17:
Ping issue is fixed in develop branch, so this will be available in next release (and it's not related to this issue).

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. This should be fixed in release.
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac