Ysurac/openmptcprouter #764: Can't get public IP address from shadowSocks

8252610, 2019-12-25:
Multiple reinstallations of VPS and routers X86-64
[image]
[image]
[image]
[image]

  - OpenMPTCProuter version: (0.53)
  - OpenMPTCProuter VPS version: ()
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-12-25:
Check if admin key is correct, and check on /var/log/daemon.log on the VPS if there is no errors.

8252610, 2019-12-25:
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4000238/daemon.log)
I'm sorry I'm not very good at reading this file. Could you help me to see if there's any mistake

8252610, 2019-12-25:
[image]
I don't think I have the wrong password

8252610, 2019-12-25:
I'm sorry to bother you with Christmas

Ysurac, 2019-12-25:
Username should be "openmptcprouter" not sure this work as "OpenMPTCProuter"

8252610, 2019-12-25:
[image]
[image]
I still haven't solved the problem. I just updated the 0.53.1 firmware

8252610, 2019-12-25:
I use Advanced settings
I didn't find the fill in form for mlvpn
[image]
[image]

Ysurac, 2019-12-25:
You checked ```/var/log/daemon.log``` on the VPS ?

8252610, 2019-12-25:
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4000606/daemon.log)
[image]

This is my latest daemon.log
I'm not very good at reading this document

Ysurac, 2019-12-25:
Shadowsocks seems unable to mount because IPv6 not enabled but it's not the real problem because this should be fixed when router can contact admin script. So the problem is why it can get a token ? And I don't see any omr-admin contact from the router...
The VPS can be contacted from the router ? Can you check ```ip r``` from the router via SSH ?

8252610, 2019-12-26:
[image]
I SSH into the router
I can enter the IP r command
Router Ping VPS IP
I can SSH VPS in the router
So the router is connected to VPS? Do you mean this?

8252610, 2019-12-26:
OK, I'll try
How can the router contact VPS?
Is SSH VPS in the router?
I can SSH into the router 
Enter IP r command
I'm outside today, and I'll give you the answer tomorrow

8252610, 2019-12-28:
I bought another VPS system, Debian 10
VPS script installed
I have reinstalled my router system
There are some problems
[image]
[image]
[image]
[image]

Ysurac, 2019-12-28:
As you are in China, I would say that Shadowsocks is blocked.
You can still check in ```/etc/shadowsocks/manager.json``` that you have same method and same key used on OMR. If it's the case, you can try v2ray obfuscation in System->OpenMPTCProuter and Advanced Parameters tab.

8252610, 2019-12-28:
I borrowed someone's IP and server key
It can be used normally. The server is Alibaba cloud in China
I can get his image. Can you tell me how to change the password  shadowsocks key  glorytun key  A Dead Simple VPN key Your MLVPN password   OpenMPTCProuter Server key
Thank you very much.

Ysurac, 2019-12-28:
Passwords are in System->OpenMPTCProuter->Wizard tab and advanced settings checkbox.

8252610, 2019-12-28:
[image]
[image]
I'm sorry, but I can't find the password

Ysurac, 2019-12-28:
In wizard page.

8252610, 2019-12-29:
Good morning
It may be a problem with the translation software
I want to change these passwords. Can I change them
[image]

8252610, 2019-12-29:
I succeeded.
v2ray obfuscation
How I operate 
1:
* Initialize image Debian 9.9
* Upgrade the code to debian10 and run the debian10 script
2：
* Initialize image Debian 10.2
* Run debian10 script directly 
* v2ray obfuscation
[image]
[image]

Ysurac, 2019-12-29:
Debian9 image is updated to Debian10 via the script, so should be quite the same.

8252610, 2019-12-30:
[image]
Previous questions
Because my VPS is 512M RAM
If VPS is 1g RAM, no problem

jfh75, 2020-01-06:
Same problem here. To sum up and add complementary information, 
1) During VPS upgrade, I FISRT had the var lock warning, stating E: Impossible d'obtenir le verrou de dpkg (/var/lib/dpkg/lock-frontend). Il est possible qu'un autre processus l'utilise.
Maybe you remember, you told me to relaunch. I did, then it went untill the end.
2) In the fisrt aborted launch, I also got plenty of warnings because ubuntu server sources are multiple
3) Hereafter in the "successfull" VPS upgrade, you may see that when compiling, there are plenty of "NO" wether library or program is present or not. Is it ok ?
[loginstallvps.txt](https://github.com/Ysurac/openmptcprouter/files/4025940/loginstallvps.txt)

Ysurac, 2020-01-06:
You should reinstall the VPS from scratch (and use Debian 10 if you don't use it for something else).
The script was run while the VPS was updating himself. Some provider seems to give access to VPS before update is finished, so you should wait before running the script (I will try to detect that on the script for next release).

jfh75, 2020-01-06:
My VPS is a separate miniPC I installed a clean ubuntu server 18.04 on on purpose.
You said I shouldn't update anything and accordingly, I didn't touched anything since. As you can see in the install log, there are 118 paquets waiting for upgrade I let unchanged :-(
I just run the scripts when a VPS update is available.
So normally, there shouldn't be any update at all, nothing is changing on this dedicated PC I haven't requested. There is no VPS supplier doing anything, just me, running the script when a new version is out ?
Does it mean a new linux install + VPS script ? But then the 118 upgradeable paquets will be updated no ?
Thank you for you help,

Ysurac, 2020-01-06:
What I mean is that an update seems to be running while you run the script, at the same time.
You can reinstall all from scratch, or try ```sudo dpkg --configure -a```. But I'm not sure what is the current state of your install...

jfh75, 2020-01-06:
For the moment, I have installed libbsd-dev and valgrind that were missing, rebooted, and relaunched the VPS install script. This time, it went through in one time
[loginstallvps2.txt](https://github.com/Ysurac/openmptcprouter/files/4026046/loginstallvps2.txt)

8252610, 2020-01-07:
Check if the port is open

8252610, 2020-01-07:
> 目前，我已经安装了缺少的libbsd-dev和valgrind，重新启动并重新启动了VPS安装脚本。这次，它一次完成了
> [loginstallvps2.txt](https://github.com/Ysurac/openmptcprouter/files/4026046/loginstallvps2.txt)

Congratulations

8252610, 2020-01-07:
> 通过脚本将Debian9映像更新为Debian10，因此应该完全相同。

> 通过脚本将Debian9映像更新为Debian10，因此应该完全相同。

> Debian9 image is updated to Debian10 via the script, so should be quite the same.

> For the moment, I have installed libbsd-dev and valgrind that were missing, rebooted, and relaunched the VPS install script. This time, it went through in one time
> [loginstallvps2.txt](https://github.com/Ysurac/openmptcprouter/files/4026046/loginstallvps2.txt)

Congratulations

8252610, 2020-01-07:
> Debian9 image is updated to Debian10 via the script, so should be quite the same.

When I use Chinese servers, I need to build a ladder for the servers to access GitHub normally. Can you tell me what service configuration to use
I'm worried that if I use shadingsocks and run the script again, it will affect VPS

jfh75, 2020-01-07:
Hello,

Today I managed to catch problem with sources at beguinning of VPS install script. I don't know why I couldn't find it in the log. Anyway here it is : 
root@vps:/home/jfh# wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh
URL transformed to HTTPS due to an HSTS policy
--2020-01-07 06:31:59--  https://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 94.23.252.192
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 39772 (39K) [application/octet-stream]
Saving to: ‘STDOUT’

-                   100%[===================>]  38,84K  --.-KB/s    in 0,001s  

2020-01-07 06:31:59 (37,5 MB/s) - written to stdout [39772/39772]

Hit:1 http://archive.ubuntu.com/ubuntu bionic InRelease
Get:2 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:3 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]    
Get:4 http://archive.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]     
Hit:5 https://repo.openmptcprouter.com stretch InRelease                       
Get:6 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [828 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-updates/main Translation-en [290 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [25.8 kB]
Get:9 http://archive.ubuntu.com/ubuntu bionic-updates/restricted Translation-en [6964 B]
Get:10 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1035 kB]
Get:11 http://archive.ubuntu.com/ubuntu bionic-updates/universe Translation-en [320 kB]
Get:12 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [9500 B]
Get:13 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse Translation-en [4540 B]
Fetched 2771 kB in 1s (2714 kB/s)                        
Reading package lists... Done
W: Target Packages (universe/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Translations (universe/i18n/Translation-fr) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target CNF (universe/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target CNF (universe/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Translations (main/i18n/Translation-fr) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target CNF (main/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target CNF (main/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Packages (universe/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Translations (universe/i18n/Translation-fr) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target CNF (universe/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target CNF (universe/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list.d/bionic-universe.list:1
W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
W: Target Translations (main/i18n/Translation-fr) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
[... cut ...]
W: Target CNF (main/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:34 and /etc/apt/sources.list.d/bionic-backports.list:1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gnupg is already the newest version (2.2.4-1ubuntu1.2).
apt-transport-https is already the newest version (1.6.12).
0 upgraded, 0 newly installed, 0 to remove and 122 not upgraded.

It is a new problem, I never got this before.
Does it help ?

Best regards
JFH

jfh75, 2020-01-07:
yesterday it went through, this morning it didn't : 
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for thread local storage (TLS) class... __thread
checking for mbedtls_cipher_setup in -lmbedcrypto... yes
checking whether mbedtls supports Cipher Feedback mode or not... ok
checking whether mbedtls supports the ARC4 stream cipher or not... ok
checking whether mbedtls supports the Blowfish block cipher or not... ok
checking whether mbedtls supports the Camellia block cipher or not... ok
checking for sodium_init in -lsodium... yes
checking for version of libsodium... yes
checking for ares_library_init in -lcares... yes
checking for C/C++ restrict keyword... __restrict
checking for netdb.h... yes
checking for library containing inet_ntop... none required
checking whether inet_ntop is declared... yes
checking for what kind of host... Linux
checking whether gcc is Clang... no
checking whether pthreads work with -pthread... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking whether more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking if -fstack-protector and -fstack-protector-all are supported.... yes
checking limits.h usability... yes
checking limits.h presence... yes
checking for limits.h... yes
checking for stdint.h... (cached) yes
checking for inttypes.h... (cached) yes
checking arpa/inet.h usability... yes
checking arpa/inet.h presence... yes
checking for arpa/inet.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking langinfo.h usability... yes
checking langinfo.h presence... yes
checking for langinfo.h... yes
checking locale.h usability... yes
checking locale.h presence... yes
checking for locale.h... yes
checking linux/tcp.h usability... yes
checking linux/tcp.h presence... yes
checking for linux/tcp.h... yes
checking netinet/tcp.h usability... yes
checking netinet/tcp.h presence... yes
checking for netinet/tcp.h... yes
checking for netdb.h... (cached) yes
checking netinet/in.h usability... yes
checking netinet/in.h presence... yes
checking for netinet/in.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for strings.h... (cached) yes
checking for unistd.h... (cached) yes
checking sys/ioctl.h usability... yes
checking sys/ioctl.h presence... yes
checking for sys/ioctl.h... yes
checking linux/random.h usability... yes
checking linux/random.h presence... yes
checking for linux/random.h... yes
checking sys/socket.h usability... yes
checking sys/socket.h presence... yes
checking for sys/socket.h... yes
checking for net/if.h... yes
checking for linux/if.h... yes
checking for linux/netfilter_ipv4.h... yes
[... 571 lines cut ...]
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1594 (1.6K) [application/octet-stream]
Saving to: '/etc/sysctl.d/90-shadowsocks.conf'

/etc/sysctl.d/90-shadowsocks.conf                           100%[=========================================================================================================================================>]   1.56K  --.-KB/s    in 0s      

2020-01-07 06:33:27 (137 MB/s) - '/etc/sysctl.d/90-shadowsocks.conf' saved [1594/1594]

--2020-01-07 06:33:27--  https://www.openmptcprouter.com/server/shadowsocks-libev-manager@.service.in
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 94.23.252.192
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|94.23.252.192|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 303 [application/octet-stream]
Saving to: '/lib/systemd/system/shadowsocks-libev-manager@.service'

/lib/systemd/system/shadowsocks-libev-manager@.service      100%[=========================================================================================================================================>]     303  --.-KB/s    in 0s      

2020-01-07 06:33:27 (15.1 MB/s) - '/lib/systemd/system/shadowsocks-libev-manager@.service' saved [303/303]

Synchronizing state of shadowsocks-libev.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable shadowsocks-libev
Install OBFS
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?

Ysurac, 2020-01-07:
As I force remove lock before running apt-get to install OBFS, I would say that again you have an update or something using apt running in background.

jfh75, 2020-01-07:
What action may I take ? The only thing I am doing is launching the install script or rebooting ?
I also did the dpkg configure -a command.

Thank you for your help
JFH

jfh75, 2020-01-07:
Can I reinstall from formatting keeping the old keys ? I use OMR 50% for remote access, If I erase the old keys, then, I am loosing access to the router.

Thank you for your help,

closed 2020-05-18 by Ysurac