Ysurac/openmptcprouter #766: openmptcp router reboot randomy

anas94b, 2019-12-27:
hi i have a problem that the router is reboo randomy after traffic pass no cpu and ram overload , testing on vmware15 and virtualbox 6.1

Ysurac, 2019-12-27:
OMR version ?
Without any log I can't say anything. I'm not able to reproduce the issue on virtualbox under Linux.

anas94b, 2019-12-28:
the log 

Sat Dec 28 02:11:40 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:11:40 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:11:41 2019 daemon.err glorytun[20556]: 172.105.71.204.65001: connected
Sat Dec 28 02:11:41 2019 daemon.info glorytun[20556]: STARTED tun0
Sat Dec 28 02:11:41 2019 daemon.err glorytun[20556]: tun write: I/O error
Sat Dec 28 02:11:41 2019 daemon.info glorytun[20556]: STOPPED tun0
Sat Dec 28 02:11:41 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:11:41 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:11:41 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:12:11 2019 daemon.info glorytun[21454]: INITIALIZED tun0
Sat Dec 28 02:12:11 2019 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Dec 28 02:12:11 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:12:11 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:12:12 2019 daemon.err glorytun[21454]: 172.105.71.204.65001: connected
Sat Dec 28 02:12:13 2019 daemon.info glorytun[21454]: STARTED tun0
Sat Dec 28 02:12:13 2019 daemon.err glorytun[21454]: tun write: I/O error
Sat Dec 28 02:12:13 2019 daemon.info glorytun[21454]: STOPPED tun0
Sat Dec 28 02:12:13 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:12:13 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:12:13 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:12:13 2019 daemon.err /usr/bin/ss-redir[15843]: remote recv: Connection reset by peer
Sat Dec 28 02:12:43 2019 daemon.info glorytun[22337]: INITIALIZED tun0
Sat Dec 28 02:12:43 2019 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Dec 28 02:12:43 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:12:43 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:12:44 2019 daemon.err glorytun[22337]: 172.105.71.204.65001: connected
Sat Dec 28 02:12:44 2019 daemon.info glorytun[22337]: STARTED tun0
Sat Dec 28 02:12:44 2019 daemon.err glorytun[22337]: tun write: I/O error
Sat Dec 28 02:12:44 2019 daemon.info glorytun[22337]: STOPPED tun0
Sat Dec 28 02:12:44 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:12:44 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:12:44 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:13:14 2019 daemon.info glorytun[23217]: INITIALIZED tun0
Sat Dec 28 02:13:14 2019 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Dec 28 02:13:14 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:13:14 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:13:15 2019 daemon.err glorytun[23217]: 172.105.71.204.65001: connected
Sat Dec 28 02:13:15 2019 daemon.info glorytun[23217]: STARTED tun0
Sat Dec 28 02:13:16 2019 daemon.err glorytun[23217]: tun write: I/O error
Sat Dec 28 02:13:16 2019 daemon.info glorytun[23217]: STOPPED tun0
Sat Dec 28 02:13:16 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:13:16 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:13:16 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:13:16 2019 daemon.err /usr/bin/ss-redir[15843]: remote recv: Connection reset by peer
Sat Dec 28 02:13:46 2019 daemon.info glorytun[24070]: INITIALIZED tun0
Sat Dec 28 02:13:46 2019 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Dec 28 02:13:46 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:13:46 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:13:47 2019 daemon.err glorytun[24070]: 172.105.71.204.65001: connected
Sat Dec 28 02:13:47 2019 daemon.info glorytun[24070]: STARTED tun0
Sat Dec 28 02:13:47 2019 daemon.err glorytun[24070]: tun write: I/O error
Sat Dec 28 02:13:47 2019 daemon.info glorytun[24070]: STOPPED tun0
Sat Dec 28 02:13:47 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:13:47 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:13:47 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Dec 28 02:14:17 2019 daemon.info glorytun[25195]: INITIALIZED tun0
Sat Dec 28 02:14:17 2019 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Dec 28 02:14:17 2019 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Dec 28 02:14:17 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Sat Dec 28 02:44:40 2019 daemon.info glorytun[12794]: STOPPED tun0
Sat Dec 28 02:44:40 2019 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Dec 28 02:44:40 2019 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Dec 28 02:44:40 2019 user.notice ModemManager: hotplug: error: parent device sysfspath not found
-----------------------------------------------------------------

openmptcprouter v0.53.1 r0+11546-a74095c68c

anas94b, 2019-12-28:
if i used with no traffic it's normal and not rebooted tested on vmware and virtual box

anas94b, 2019-12-28:
kernel log : 

[    0.000000] Linux version 4.19.84 (circleci@default-c95d1829-bdad-44a8-bdfd-251b28ad7da7) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r0+11546-a74095c68c)) #0 SMP Tue Nov 19 13:59:03 2019
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=5926140f-02 rootfstype=ext4 rootwait console=tty0 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000061ceffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000061cf0000-0x0000000061cfffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 23d9001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 1156714485287 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 2494.334 MHz processor
[    0.003265] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003268] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003271] last_pfn = 0x61cf0 max_arch_pfn = 0x400000000
[    0.003283] MTRR default type: uncachable
[    0.003284] MTRR variable ranges disabled:
[    0.003284] Disabled
[    0.003285] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.003288] CPU MTRRs all blank - virtualized system.
[    0.003291] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.003357] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.003401] BRK [0x02601000, 0x02601fff] PGTABLE
[    0.003403] BRK [0x02602000, 0x02602fff] PGTABLE
[    0.003405] BRK [0x02603000, 0x02603fff] PGTABLE
[    0.003431] BRK [0x02604000, 0x02604fff] PGTABLE
[    0.003542] BRK [0x02605000, 0x02605fff] PGTABLE
[    0.003586] ACPI: Early table checksum verification disabled
[    0.003608] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.003612] ACPI: XSDT 0x0000000061CF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.003618] ACPI: FACP 0x0000000061CF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.003624] ACPI: DSDT 0x0000000061CF0470 002325 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.003629] ACPI: FACS 0x0000000061CF0200 000040
[    0.003632] ACPI: FACS 0x0000000061CF0200 000040
[    0.003635] ACPI: APIC 0x0000000061CF0240 00005C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.003639] ACPI: SSDT 0x0000000061CF02A0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.003649] ACPI: Local APIC address 0xfee00000
[    0.004334] Zone ranges:
[    0.004339]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004341]   DMA32    [mem 0x0000000001000000-0x0000000061ceffff]
[    0.004342]   Normal   empty
[    0.004343] Movable zone start for each node
[    0.004344] Early memory node ranges
[    0.004345]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.004346]   node   0: [mem 0x0000000000100000-0x0000000061ceffff]
[    0.004772] Reserved but unavailable: 98 pages
[    0.004775] Initmem setup node 0 [mem 0x0000000000001000-0x0000000061ceffff]
[    0.004778] On node 0 totalpages: 400526
[    0.004779]   DMA zone: 64 pages used for memmap
[    0.004779]   DMA zone: 21 pages reserved
[    0.004780]   DMA zone: 3998 pages, LIFO batch:0
[    0.004850]   DMA32 zone: 6196 pages used for memmap
[    0.004851]   DMA32 zone: 396528 pages, LIFO batch:63
[    0.015317] ACPI: PM-Timer IO Port: 0x4008
[    0.015322] ACPI: Local APIC address 0xfee00000
[... 492 lines cut ...]
[24413.473864] __mptcp_init4_subsockets: token 0x3ae3b7f7 bind() to 185.138.121.236 index 7 failed, error -99
[24413.584812] __mptcp_init4_subsockets: token 0x3ff98a69 bind() to 185.138.121.236 index 7 failed, error -99
[24413.642353] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[24413.647115] 8021q: adding VLAN 0 to HW filter on device eth2
[24413.673043] __mptcp_init4_subsockets: token 0xf34edf62 bind() to 185.138.121.236 index 7 failed, error -99
[24413.711147] __mptcp_init4_subsockets: token 0x6e8d1a61 bind() to 185.138.121.236 index 7 failed, error -99
[24413.725706] __mptcp_init4_subsockets: token 0xc2b43d7a bind() to 185.138.121.236 index 7 failed, error -99
[24413.743474] __mptcp_init4_subsockets: token 0x383c7aa4 bind() to 185.138.121.236 index 7 failed, error -99
[24413.800005] __mptcp_init4_subsockets: token 0x8d9a469c bind() to 185.138.121.236 index 7 failed, error -99
[... cut ...]
[26491.363388] __mptcp_init4_subsockets: token 0x27efe3b5 bind() to 185.138.121.236 index 7 failed, error -99
[26491.363910] __mptcp_init4_subsockets: token 0x6ca86018 bind() to 185.138.121.236 index 7 failed, error -99
[26491.364434] __mptcp_init4_subsockets: token 0xe147c638 bind() to 185.138.121.236 index 7 failed, error -99
[26491.498653] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[26491.504100] 8021q: adding VLAN 0 to HW filter on device eth2
[26493.523074] e1000: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX

Ysurac, 2019-12-28:
Many errors with glorytun, MPTCP and modemmanager. A conflict of modemmanager that try to use tun0 already use by glorytun ?
For the reboot issue, what is the host platform ? Windows, MacOS, Linux ?

anas94b, 2019-12-28:
Windowes 10

anas94b, 2019-12-28:
after looking hard this the problem apper before restart 

Kernel Panic - not syncing: Fatal exception in interrupt

Ysurac, 2019-12-28:
I would need the log before that.
If you use MacVLAN try to not use it.

I never tested OpenMPTCProuter under Windows, can be some problems with MPTCP...

anas94b, 2019-12-28:
i am using a vlan on my intel ethernet and bridged with virualbox .
i need the best way for high traffic using

anas94b, 2019-12-28:
A recored the problem .. ser the problem below

[image]
[image]
[image]
[image]
[image]

Ysurac, 2019-12-28:
Idle problem first and then seems a nat problem.
Firewall and hibernate mode are disabled on host ? I can't really say where is the problem, I'm not able to reproduce the issue on Linux.

anas94b, 2019-12-28:
I will change to linux and check 

Can you tall me how to add i vlan ? 

Because i must use the vlan for wan1 .. wan2 .. lan

anas94b, 2019-12-28:
same problem after high traffic it's restart and make kernel down

Ysurac, 2019-12-28:
Can be a hardware problem

anas94b, 2019-12-28:
i will change the pc also

anas94b, 2019-12-29:
same i have almost try every think same problem 

the problem if i used for my pc it's work with no problem but if i used for a small network @50 mbit .. i get a problem 

i want argent like this system for my work if you know i paid service like this no problem

anas94b, 2019-12-29:
after testing and testing it's work normaly with high traffic just for let's say 10 min and after that it's restart 
and if no traffic it's work without restart for long time the problem with kernal i think 

i want to use old ver ? how shall it's work with vps script

Ysurac, 2019-12-29:
50Mb/s is not really a high traffic.
Can you test ```omr-test-speed``` via SSH on the router ? This will test if it's a nat issue or something else.

anas94b, 2019-12-29:
i will test now

anas94b, 2019-12-29:
[image]

anas94b, 2019-12-29:
[image]

no problem i think the only problem when there is a high number of  connection  >>

anas94b, 2019-12-29:
[image]

anas94b, 2019-12-29:
[image]
finshed no problem !

Ysurac, 2019-12-29:
You can try "Software flow offloading" in Network->firewall
Else I have no real idea...
What do you call "high number of connection" ?

anas94b, 2019-12-29:
can try "Software flow offloading" in Network->firewall

i will try 

high number of connection -- || a lot of users about 100 users using the internet

anas94b, 2019-12-29:
i have like this  problem 

https://github.com/Ysurac/openmptcprouter/issues/270

and i do not think it's fixed ?

Ysurac, 2019-12-29:
nlbwmon was installed, it's not installed by default.
You can try to remove xt_ndpi : ```rmmod xt_ndpi``` via SSH on the router (it's used to detect traffic).

anas94b, 2019-12-29:
> nlbwmon was installed, it's not installed by default.
> You can try to remove xt_ndpi : `rmmod xt_ndpi` via SSH on the router (it's used to detect traffic).

after remove it .. the reboot also happen !

Ysurac, 2019-12-29:
No idea other than try to use a dedicated server or at least not under Windows.

anas94b, 2019-12-29:
Tested direct on a pc with image ext4 ..same problem

Ysurac, 2019-12-29:
Strange... I'm not able to reproduce the issue and first report on dedicated server.
Each time with a trace about nat ?

anas94b, 2019-12-29:
I can provide you a public access to every thing if you want to see.. 
I do not think it's a nat because it's work for some time and randomy reboot

Ysurac, 2019-12-29:
You can try in Network->MPTCP to use another TCP Congestion Control like olia and then cubic. If still no change you can try to disable MPTCP to check if it's related to MPTCP.
If it's working without MPTCP you can try with MPTCP and another scheduler.

anas94b, 2019-12-30:
I will check soon .. can you provide me an old version( vps script and ex4 image )  with different kernal i think the problem from it

Ysurac, 2019-12-30:
Check on git but I don't support old release.

anas94b, 2019-12-30:
I could not find an old version for vps script

Ysurac, 2019-12-30:
here: https://github.com/Ysurac/openmptcprouter-vps

closed 2020-05-18 by Ysurac