Ysurac/openmptcprouter #768: omr-admin status failed VPS

runfortheapples, 2019-12-28:
## Expected Behavior

 omr-admin service on vps status to be active.

## Actual Behavior
omr-admin service on vps current status is failed:
[image]

OenMPTCProuter can't contact Server Admin Script:
[image]

## Steps to Reproduce the Problem

  1. upgrade openmptcprouter from 0.53.0 to 0.53.1
  2. use vps script to upgrade server from 0.1004 to 0.1005

## Specifications

  - OpenMPTCProuter version: 0.53.1
  - OpenMPTCProuter VPS version: 0.1005 Ubuntu 18.04
  - OpenMPTCProuter platform: RPI4(64-bit)

Screenshot of syslog:
[image]

Ysurac, 2019-12-28:
```pip3 install pyjwt``` should solve the issue, but this should have be installed via the script.

runfortheapples, 2019-12-28:
[image]

Seems to be already installed.

Ysurac, 2019-12-28:
Strange... I will check on a Ubuntu 18.04 if I can reproduce the issue...

Ysurac, 2019-12-28:
I'm not able to reproduce the issue on Ubuntu 18.04 from a fresh install with latest script.

runfortheapples, 2019-12-28:
Ran a fresh instance of ubuntu18.04 and had the same error. Got the service to start by doing `pip3 install jwt` but still had the error that the openmptcprouter couldn't contact the VPS. 
In the end I started a fresh instance of debian9 and ran the script. 
Router can now contact the VPS:
[image]

Thank you for your time @Ysurac

closed 2019-12-28 by runfortheapples