Ysurac/openmptcprouter #769: omr-bypass for all UDP traffic

Bestle, 2019-12-28:
How do I enable omr-bypass for all UDP traffic? It doesn't come up in the list of protocols.

Ysurac, 2019-12-29:
You can't, but you can disable VPN and use shadowsocks only for TCP traffic.

max4186, 2020-01-02:
I do this actively, disabling the VPN and using Shadowsocks only does the trick.

Bestle, 2020-01-03:
Yeah I'm having the issue you had @max4186. When playing games (UDP), I get crazy ping fluctuations when someone else on the network is downloading or watching YouTube.

The bandwidth they are using is nowhere near my box and neither the router nor VPS are constrained for resources, it's very odd.

max4186, 2020-01-03:
Mine was worst when watching any kind of video, like 2000ms ping. Luckily, just disabling the VPN has solved that problem entirely :)

Bestle, 2020-01-03:
Hopefully @Ysurac can fix the route cause, but will try that workaround in the meantime.

Ysurac, 2020-01-03:
There is SQM and you can choose between multiple VPN.

Bestle, 2020-01-04:
I get the same issue with other VPN. I currently use 4G connections with variable speeds so can't really use SQM.

Disabling the VPN completely fixes the issue, but surely that means it isn't an issue that can be fixed with SQM?

Bestle, 2020-01-06:
@Ysurac Which WAN is selected for traffic if VPN is off?

max4186, 2020-01-06:
The master multipath wan, so probably wan1 @Bestle

Bestle, 2020-01-09:
So with VPN disabled, UDP (game) traffic is flawless with no fluctuations in ping when watching a youtube video at the same time.

closed 2020-05-18 by Ysurac