Ysurac/openmptcprouter #770: DNS config issue

couili, 2019-12-30:
## Expected Behavior
Hello,

Since the last update, I can't enter any more DNS servers in the general parameters, under "DNS transmission".
I have an error "Expecting: valid hostname".
In previous versions of OMPTCPRouter, a simple IP address was sufficient.
Sorry for my English, I'm French.

Thank you for your help

## Specifications

  - OpenMPTCProuter version: v0.53.1 r0+11546-a74095c68c 
  - OpenMPTCProuter VPS version: 0.1005 4.19.80-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-12-30:
True it's a bug in LuCi master branch from OpenWrt. Not yet fixed. I will fix it for next release.

couili, 2019-12-30:
Ok, j'espère qu'une MAJ arrivera rapidement. Merci !

closed 2019-12-31 by couili