Ysurac/openmptcprouter #771: Configuration IPv6

iamfreaky, 2019-12-30:
## Expected Behavior
IPv6 /64 Subnet on VPS ist available. 
When i turn on IPv6 Support, Clients Should get a IPv6 Address.

## Actual Behavior
Clients doesnt Get Any IPv6 Address. 

## Steps to Reproduce the Problem

  1. Activate IPv6 in Gateway Settings

## Specifications

  - OpenMPTCProuter version: (v0.53.1 r0+11546-a74095c68c)
  - OpenMPTCProuter VPS version: (0.1005)
  - OpenMPTCProuter platform: (x86_64 ESXi OVA)

OpenMPTCPRouter has IPv6 Connection itself and can reach IPv6 Addresses but nothing behind the router because of the lack of an IPv6 Address. Clients behind Router are Debian 10 and they have IPv6 enabled. Clients are using eth1 on the Router side to get into the internet. 

Log of ip -6 VPS
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a03:XXXX:XXXX:XXXX:ebf3/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::d479:9aff:fe5d:ebf3/64 scope link
       valid_lft forever preferred_lft forever
3: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::c646:abb3:a810:7e25/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
4: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::46db:aaa9:a316:9a26/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::4a6f:9eff:835:511f/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::f36d:c5c4:bd76:d86b/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1280 state UNKNOWN qlen 500
    inet6 fe80::b3ea:5f2:1593:1699/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
10: omr-6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 state UNKNOWN qlen 1000
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/64 scope link
       valid_lft forever preferred_lft forever
12: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::a159:7099:3736:2639/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

Log of Client
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000                                                             inet6 ::1/128 scope host                                                                                                   valid_lft forever preferred_lft forever                                                                          2: ens224: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000                                                    inet6 fdbb:9341:ecdf:0:505e:1534:c9a1:cd6/64 scope global temporary dynamic                                                valid_lft 601083sec preferred_lft 82221sec                                                                           inet6 fdbb:9341:ecdf:0:250:56ff:fea9:5be8/64 scope global mngtmpaddr noprefixroute                                         valid_lft forever preferred_lft forever                                                                              inet6 fe80::250:56ff:fea9:5be8/64 scope link noprefixroute                                                                 valid_lft forever preferred_lft forever  

Log of Router
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 state UP qlen 1000
    inet6 fdbb:9341:ecdf::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::250:56ff:fea9:b320/64 scope link
       valid_lft forever preferred_lft forever
6: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fdbb:9341:ecdf::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::250:56ff:fea9:2692/64 scope link
       valid_lft forever preferred_lft forever
8: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1492 state UP qlen 100
    inet6 fe80::4470:97ff:feda:5aa6/64 scope link
       valid_lft forever preferred_lft forever
9: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1492 state UP qlen 100
    inet6 fe80::c8bd:90ff:fec3:2319/64 scope link
       valid_lft forever preferred_lft forever
15: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 1000
    inet6 fe80::2053:b2b9:4612:3256/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
17: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 state UNKNOWN qlen 1000
    inet6 fe80::a00:2/128 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever

Ysurac, 2019-12-30:
Check in network->interface->LAN on DHCP server tab and IPv6 Settings tab that you have "Router Advertisement-Service" set as "Server mode" and "Always announce default router" enabled.

iamfreaky, 2019-12-30:
@Ysurac Sorry still doesnt work. Needs there to be any other Configuration ? 
[image]

Ysurac, 2019-12-30:
You should let IPv6 assignment length to the default 60

iamfreaky, 2019-12-30:
@Ysurac Thank you, it looks like its working now. Only there is no DHCPv6 Lease vor my pc. In the Overview of the Router. It is then still possible to open a Port for the Client ? Like in IPv4 ?

Ysurac, 2019-12-30:
Not yet possible to open port for IPv6 client.
I now have a VPS with a /48, so some work will be done on IPv6 soon.

iamfreaky, 2019-12-30:
@Ysurac Thats sounds great, hope that there will be fast progress. 

For now i think this can be closed IPv6 Connection is working.

closed 2019-12-30 by iamfreaky