Ysurac/openmptcprouter #774: Can't update VPS

couili, 2020-01-01:
## Expected Behavior

 Hello,

I can't update my VPS today, i have this error :

usermod: no changes
Created symlink /etc/systemd/system/multi-user.target.wants/mlvpn@mlvpn0.service -> /lib/systemd/system/mlvpn@.service.
install mlvpn done
Install OpenVPN
Reading package lists... Done
Building dependency tree       
Reading state information... Done
easy-rsa is already the newest version (3.0.6-1).
openvpn is already the newest version (2.4.7-1).
The following packages were automatically installed and are no longer required:
  libevent-core-2.1-6 libevent-pthreads-2.1-6 libopts25 sntp
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 45 not upgraded.
mv: cannot move '/etc/openvpn/server/ca.crt' to '/etc/openvpn/ca/pki/ca.crt': No such file or directory

## Steps to Reproduce the Problem

  1. Launch script
  2. Wait

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1005
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-01-01:
Should be fixed now, run the script again.

couili, 2020-01-01:
Yes, Thank You !

closed 2020-01-01 by couili