Ysurac/openmptcprouter #775: Can't Update my VPS

Randdrick, 2020-01-01:
Hello,

I can't update my VPS today, i have this error :

install mlvpn done
Install OpenVPN
Reading package lists... Done
Building dependency tree
Reading state information... Done
easy-rsa is already the newest version (3.0.6-1).
openvpn is already the newest version (2.4.7-1).
The following packages were automatically installed and are no longer required:
  g++-6 guile-2.0-libs irqbalance libbind9-140 libdbus-glib-1-2 libdns162
  libicu57 libirs141 libisc160 libisccc140 libisccfg140 liblwres141
  libmbedcrypto0 libmbedtls10 libnuma1 libperl5.24 libpython3.5
  libpython3.5-dev libpython3.5-minimal libpython3.5-stdlib libstdc++-6-dev
  linux-image-4.9.0-11-amd64 linux-image-4.9.0-8-amd64 python3-pyasn1
  python3.5 python3.5-dev python3.5-minimal rsync runit-helper tcpd
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
cp: cannot create regular file '/etc/openvpn/ca/pki/ca.crt': No such file or directory
root@vps-42566:~#

## Steps to Reproduce the Problem
Launch script
Wait

## Specifications
OpenMPTCProuter version: 0.52.3
OpenMPTCProuter VPS version: 0.1005
OpenMPTCProuter platform: RP3 64

Ysurac, 2020-01-01:
Same as #774, fixed.

closed 2020-01-01 by Ysurac

Randdrick, 2020-01-01:
It's not fix for me. it's beacause i open this issue

Ysurac, 2020-01-01:
Download and run the script again, there is not more use of "cp", so it's an old release.