Ysurac/openmptcprouter #777: Problème IPv6

Slivikkus, 2020-01-01:
## Actual Behavior
[image]
J'ai enfin réussi à activé l'IPv6 mais elle n'a pas l'air active faut t-il configurer quelques chose en particulier pour la rendre active ?

## Specifications

  - OpenMPTCProuter version: (0.53.2)
  - OpenMPTCProuter VPS version: (1000.1)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-01-01:
Comment ça pas active ? Il faut que ça soit activé dans le wizard et pour distribuer les IPs il faut activer dans réseau->interface->lan puis DHCP et IPv6.

Slivikkus, 2020-01-01:
L'IPv6 est activé et bien détectée mais je peut pas faire en sorte que l'IPv4 sois remplacée par l'IPv6 ?

Ysurac, 2020-01-01:
Non, mais c'est juste de l'affichage. L'adresse IPv6 publique est utilisée pour tout le traffic IPv6.

Slivikkus, 2020-01-01:
[image]
J'obtient ça quand je vais sur https://ip.lafibre.info

Ysurac, 2020-01-01:
Depuis quel OS et navigateur ?

Slivikkus, 2020-01-01:
Windows 10 et Brave (basé sous Chromium)

Ysurac, 2020-01-01:
Je ne connais pas windows, mais il y a peut être une options privilégiant IPv4 à IPv6 qui est activée.

Slivikkus, 2020-01-01:
Sur mon téléphone je n'ai pas d'IPv6 (sauf local), je ne peux pas aller sur un site avec IPv6 seulement mais j'en ai tout de même une 🤔🤔🤔

Slivikkus, 2020-01-01:
[image]
Et encore je suis en sans fil le réseau RJ45 fonctionne mal sauf pour mes serveurs

Ysurac, 2020-01-01:
Si tu as activé le "Router Advertisement-Service" et "Always announce default router" dans Réseau->Interface Lan tu dois avoir des IPv6 privées qui sont distribuées aux clients le supportant.
Il faut faire attention de pas avoir autre chose distribuant une IPv6 (type freebox) sur le réseau.

Slivikkus, 2020-01-01:
Non c'est que IPv4 pour l'instant au niveaux des box, c'est possible de pouvoir avoir l'IPv6 publique donné par le VPS OVH disponible pour tout les équipements (qui le supportent) ?

Ysurac, 2020-01-01:
Par défaut, l'IPv6 publique du VPS est utilisée pour la sortie en IPv6 de tous les clients, une IPv6 privée étant sinon attribuée en local à chaque équipement. Ensuite la préférence IPv6 au lieu de IPv4 dépend du système d'exploitation et éventuellement logiciel utilisé.

Slivikkus, 2020-01-01:
[image]
Les paramètres sont corrects ? (de base il y avais le "Router Advertisement-Service" et "Always announce default router" d'activé, et IPv6 détectée mais là non il faut que je réactive "Router Advertisement-Service" et "Always announce default router" ?)

Ysurac, 2020-01-01:
oui active le router advertisement et always announce default router, tu dois avoir une IPv6 en "fdxx" donnée aux clients.

Slivikkus, 2020-01-01:
[image]
Comme ceci c'est bon ?

Ysurac, 2020-01-01:
oui

Slivikkus, 2020-01-01:
D'ailleurs je ne comprend pas pourquoi mon ordinateur et mon raspberry pi n'arrive pas à ce connecter au routeur 🤔🤔

Slivikkus, 2020-01-01:
[image]

Ysurac, 2020-01-01:
pas d'IPv4 non plus ? le DHCP IPv4 est désactivé ?

Slivikkus, 2020-01-01:
Non pourtant j'ai rien touché sur l'IPv4 y'a que mon routeur WiFi Xiaomi (qui attribue des IP propre au équipements connectés sur le routeur wifi)

Ysurac, 2020-01-01:
alors un soucis de VLAN ou de réseau.

Slivikkus, 2020-01-01:
Comment je peut y remédier ?

Ysurac, 2020-01-01:
Si tu sais pas ce qu'est un VLAN c'est que tu en as surement pas, pour le réseau faut vérifier les branchements.

Slivikkus, 2020-01-01:
Réseau aucun pb puisque mes serveurs sont en ligne avec une IPv4 🤔🤔

Ysurac, 2020-01-01:
Je parle du réseau avec OpenMPTCProuter. Sinon voir si il n'y a pas un équipement bloquant le DHCP.

Slivikkus, 2020-01-01:
J'ai rien qui bloque l'attribution des IP 🤔🤔 (misère)

Slivikkus, 2020-01-01:
[image]
Sur mon serveur j'ai pas d'IPv6 détectés sauf sur les VM linux

Ysurac, 2020-01-01:
Je peux rien dire à propos de Windows, je connais pas du tout.

Slivikkus, 2020-01-01:
Par défaut sur Windows les attributions ce font automatiquement comme sur Linux mais y'a de l'IPv6 sans en avoir bizarre, ça en fait des problème 🤔 

- Pas d'attribution d'IP en ethernet sauf pour les serveurs
- L'IPv6 à l'air de fonctionner partiellement

Peut être un problème de configuration quelques part ?

Slivikkus, 2020-01-01:
J'ai trouver le problème quand j'active le DHCPv6 l'ethernet cesse de fonctionner mais une fois l'IPv6 inactif ça refonctionne ainsi que l'attribution du DHCPv4

Ysurac, 2020-01-01:
Essaye uniquement "Router Advertisement-Service" et "always announce default router"

Slivikkus, 2020-01-01:
[image]
Comme ça ?

Ysurac, 2020-01-01:
oui

Slivikkus, 2020-01-01:
D'accord

Slivikkus, 2020-01-01:
Pour l'instant ça à l'air d'aller mais la priorité est toujours mis sur l'IPv4 normalement ça devrait automatiquement prendre la relève non ? 🤔

Slivikkus, 2020-01-01:
[image]
Bizarre sur quelques site avec IPv6 ça me fait ça

Slivikkus, 2020-01-01:
C'est quoi le Global network options >IPv6 ULA-Prefix par défaut ? Je me suis amusé à le modif mais ça n'a pas aimer on dirais

closed 2020-01-04 by Slivikkus