Ysurac/openmptcprouter #778: Using Shadowsocks without speed increase (wan2)

max4186, 2020-01-02:
Is there any way you can effectively use Shadowsocks to proxy your connection without multipathing onto WAN2 to specific IP addresses rather than client/network wide? I get really bad routing to something that improves massively using the VPN but I don't want it to use my bonded bandwidth as I have a limit on it.

max4186, 2020-01-02:
Also, how can I verify my VPN is actually working? I've been dealing with an issue where anything I bypass only gets to 4.8Mbps solid bandwidth usage after the most recent upgrade, I have the VPN on but when I run a traceroute its going out via wan1 which to me means its not working....

Ysurac, 2020-01-02:
If you use bypass, traffic output to the selected wan directly, so this doesn't use shadowsocks or VPN.
It's not possible to output via shadowsocks without aggregation.

max4186, 2020-01-02:
Unfortunately, this still causes issues. Specifically on a certain device, whenever I use bypass and then connect to an international network it maxes out at 4.8Mbps. I tried fresh installs of both RPI and VPS and it has fixed the VPN itself but not the bypass speed. Note when I do bypass and then try on my mobile device, it maxes out wan1 as expected, no 4.8Mbps limit.

This is a new issue as of the most recent update, up until now I haven't had this problem - note I have also ran the RPI firmware update command.

Ysurac, 2020-01-02:
You can try with ```omr-test-speed wan2``` via SSH on the router to check the max speed possible via OpenMPTCProuter.

max4186, 2020-01-02:
I do get maximum speed on both wan1 and wan2, my issue is that when I'm using bypass on a specific device on my network to ANY international hosted server/download - my speed solid is 4.8Mbps but when I do the same on a mobile device on same network, I get max wan1 speed which is expected.

Any ideas on what I could check to begin to diagnose? Note as I mentioned, this has only started as of the most recent update and I have tried fresh installs of both VPS & router.

Ysurac, 2020-01-02:
I don't think I understand...
You use OMR-Bypass:
* for one device to the external server you get 4.8Mbps
* for another device that use wifi this work to same external server

or the problem is with any device that use omr-bypass and the mobile device is not related at all ?

max4186, 2020-01-02:
One device using OMR-Bypass on wan1 I get 4.8Mbps, on another device to the exact same server I get max wan1 speed (20Mbps). It's only to international locations, if I do it within EU (low latency places) then I get max wan1 speed on the effected device.

Ysurac, 2020-01-02:
I would say there is a problem with the device that have 4.8Mbps and it's not related to OpenMPTCProuter as it's work on another device.

max4186, 2020-01-02:
I'd say same thing other than when I use it WITHOUT aggregation, I max both ports to the end server which means to me it's something to do with OpenMPTCP but I will continue to investigate....

Ysurac, 2020-01-02:
If you have SQM enabled it's possible.

max4186, 2020-01-02:
Unfortunately, no SQM enabled either.

max4186, 2020-01-02:
It turns out, I'm only getting 4~Mbps even through bonding - I never realized it was automatically overflowing to wan2:

Inbound:	3.40 Mbit/s
(435.16 KB/s)

What would cause it to overflow to wan2 when 4Mbps is no where near the maximum throughput?

max4186, 2020-01-02:
Is there an easy way to install the 0.52 VPS version? The install script is a .sh that hasn't been updated in 6 months so I imagine it must pull from something else you have updated?

max4186, 2020-01-03:
This can be closed, I now fixed the issue - it was a fault on my PC which I fixed with a TCP optimisation tool, sorry for the false alarm and wasting of time.

closed 2020-01-03 by Ysurac