Ysurac/openmptcprouter #779: OMR-Bypass not working

Th0rGer, 2020-01-02:
I set the OMR_Bypass yesterday and it worked the evening and the morning. Now Netflix began to block me again. To avoid the DNS problem i set up OMR-Bypass for Netflix and DNS since im running a pihole in my intern network. Netflix and DNS should go out through wan1. 

## Expected Behavior

Run Netflix without problems.

## Actual Behavior

Netflix bring the Proxy Warning

## Steps to Reproduce the Problem

  1. Turn on OMR-Bypass for Netflix and DNS
  

## Specifications

  - OpenMPTCProuter version: v0.53.2 
  - OpenMPTCProuter VPS version:  0.1006 
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-01-02:
This can't work with a custom DNS.

Th0rGer, 2020-01-02:
so i have to Install pihole on the vps?

Ysurac, 2020-01-02:
Yes or redirect to pihole in Network->DHCP & DNS and use OpenMPTCProuter as DNS on your client.

Th0rGer, 2020-01-02:
so i set up a static route to my other network (in which pihole is running) now it says the vpn is down? What does it mean?

Ysurac, 2020-01-02:
Because your route is not working correctly or conflict with vpn route. You may have to add an interface to OpenMPTCProuter that is in this other network.

Th0rGer, 2020-01-02:
So i have ma Lan network (192.168.1.0/24) in this network all of my stuff is placed. 
From there the wan port on the router is directly connected to the lan port of the OpenMPTCProuter which has the network 192.168.100.0/24. How can i add a second interface?

Th0rGer, 2020-01-03:
now i set up pihole on the vps and i cant connect to the web interface

closed 2020-01-03 by Th0rGer