Ysurac/openmptcprouter #781: DHCPV6 service  back already desable

flimo44, 2020-01-02:
## Expected Behavior

DHCPV6 service don't stay enable 

## Actual Behavior

## Steps to Reproduce the Problem

  1. Enable service
  2.  Backup and Apply : It's OK a moment 
  3.  After a moment come back to desabled

## Specifications

  - OpenMPTCProuter version:  0.53.2
  - OpenMPTCProuter VPS version: 1006
  - OpenMPTCProuter platform: x86_64

flimo44, 2020-01-04:
Router crash and reboot this night at 4h18 and Dhcpip6-service is again desabled.
I don't find error in my syslog why router routed.
I think router crash or reboot every day ( about once a day) . My router run on synology ds918+ VM.
I try reboot manualy router and  Dhscpv6-service stay enabled after restarted.

Ysurac, 2020-05-18:
This will be solved in 0.55. If not, you are free to reopen this issue.

closed 2020-05-18 by Ysurac