Ysurac/openmptcprouter #783: Since update router to 0.53.2 and vps to 0.1007 the network schema doesn't display

xillibit, 2020-01-03:
## Expected Behavior

The network schema should be displayed

## Actual Behavior

The network schema doesn't displayed

[image]

The system log, the ip of VPS has been repalced by IPVPS : https://gist.github.com/xillibit/90045f507670c3335d8e119f296a7deb

## Steps to Reproduce the Problem

  1. Update VPS to 0.1007
  2. Update router to 0.53.2
  
## Specifications

  - OpenMPTCProuter version: 0.53.2
  - OpenMPTCProuter VPS version: 0.1007
  - OpenMPTCProuter platform: RPI4 aarch64

tboome33, 2020-01-05:
Bonjour et bonne année, même problème ici :-(
Merci pour ce formidable travail
RG

Ysurac, 2020-01-06:
Should be better with 0.53.3

xillibit, 2020-01-06:
J'ai mis à jour le routeur en 0.53.3 et ensuite le VPS en 0.1008 mais le problème est toujours là : 

[image]

tboome33, 2020-01-06:
Bonsoir, idem toujours le même problème :-(

8252610, 2020-01-15:
> J'ai mis à jour le routeur en 0.53.3 et ensuite le VPS en 0.1008 mais le problème est toujours là :
> 
> [image]
I'm just like you

stanf69, 2020-02-03:
Same problem here since upgrade to 53.3.
I've tried factory reset and the NETWORK STATUS schema is still not displayed. What can I do ?

Ysurac, 2020-02-07:
This happen only when there is an issue, and a check time out. This will be fixed in next update.

xillibit, 2020-02-21:
Do-you know when is planned your next update of OMR ?

Ysurac, 2020-02-21:
When it will be ready, this should be soon.

xillibit, 2020-03-01:
Since the update of VPS to 0.1012 and openmptpcrouteur to v0.54rc7 r0+12300-fc65fae304 the network status is diplaying now

closed 2020-03-14 by xillibit