Ysurac/openmptcprouter #784: No Server ping response after 1 second

8252610, 2020-01-04:
I use Huawei USB 4G router
The system displays "no server Ping response after 1 second"
I change another SIM card。 No problem.
If there is a problem with the SIM card, it is OK to use Huawei 4G router to pass RJ45
It's hard to describe
I use Baidu translation software
[image]
[image]

## Specifications

  - OpenMPTCProuter version: (0.53.2)
  - OpenMPTCProuter VPS version: (1005)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-01-04:
The message is only an info, if it's working than you have no problem (on some config I can't bind to the interface and this display this warning).

8252610, 2020-01-04:
thank you  I will continue to observe 
How do you know that they have successfully linked

8252610, 2020-01-04:
I am installing vps1007 And then it's testing

Ysurac, 2020-01-04:
VPS and router are green with VPS IP displayed, so all should be ok.

8252610, 2020-01-04:
I see. The picture I sent before is OK

8252610, 2020-01-04:
Oh, I see

xabix, 2020-01-04:
Hello,

I have quite some issues with stability since a few weeks.

1/ Is there an easy way to do a factory reset so I can reconfigure it all?
2/ now in 0.52.3 how to configure the master interface ?
3/ I am looking to add Pi-hole on the VPS, so nice to have discovered that there is a script for that :)? THANK YOU

XabIX
[image]

8252610, 2020-01-05:
In the beginning, router and VPS were green
But in a few minutes, it's going to report an error
[image]
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS1, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty31, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty31
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty31, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: add serial interface ttyI34: event processed
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI34, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty21, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty21
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty21, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty5
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty5, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: add serial interface ttyI35: event processed
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI35, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty11, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty11
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty11, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty58, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty58
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty58, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: add serial interface ttyI36: event processed
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI36, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty3, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty3
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty3, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty48, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty48
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty48, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: add serial interface ttyI37: event processed
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI37, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS8, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS8
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS8, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: add serial interface ttyI38: event processed
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty38, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty38
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI38, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty38, subsystem=tty
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan  5 09:29:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Sun Jan  5 09:36:24 2020 user.notice omr-tracker: Can't create route to 80.67.169.12 via 192.168.4.1 dev eth5 src 192.168.4.103. waiting...
Sun Jan  5 09:36:25 2020 daemon.err /usr/bin/ss-local[20158]: getpeername: Socket not connected
Sun Jan  5 09:36:34 2020 daemon.err /usr/bin/ss-local[20158]: getpeername: Socket not connected
Sun Jan  5 09:36:42 2020 daemon.err /usr/bin/ss-local[20158]: getpeername: Socket not connected
Sun Jan  5 09:36:43 2020 daemon.err /usr/bin/ss-local[20158]: getpeername: Socket not connected

Ysurac, 2020-01-05:
Remove all interfaces not used or disable MPTCP on them via Network->MPTCP.
If there is no change after that, I will need the result of ```ip r``` again when it's failing.

8252610, 2020-01-05:
After restarting the router, you can access the Internet
After a few minutes, no access to the Internet
[image]
[image]

Ysurac, 2020-01-05:
Server route is not added. I think there is a problem with omr-tracker, you can try to put IP from China like baidu.com in Services->omr-tracker (must answer to ping in default parameter section and omrvpn, must answer to http request in ShadowSocks tracker Settings).

8252610, 2020-01-05:
I'm sorry I made a big mistake
I'm really sorry
I found a 4G router with the wrong SIM card   192.168.4.1
So I deleted what I just said
I'm testing two 4G routers
They all have access to the Internet
I will add 114.114.114.114
Some DNS in China

8252610, 2020-01-07:
I don't know if my settings are correct
It still appears. The first few minutes are normal, and then it is disconnected
In my test, there are also more than ten hours of normal operation
Most of the time, it's still disconnected
I tried to shut down three trackers
Disconnection will also occur
I have only one solution
If one of the USB 4G routers is normal, it will not be disconnected
[image]
[image]
[image]

taojiji, 2020-02-02:
How to install alicloud server successfully，I've installed it several times, and I've always failed，2045321577@qq.com

closed 2020-05-18 by Ysurac