Ysurac/openmptcprouter #785: Impossible de contacter le scrip administrateur (no token=

jfh75, 2020-01-04:
## Expected Behavior

Shadowsoscks should contact the administrator script

## Actual Behavior
With new version 53.2, shadowsocks displays impossible to contact administrator script (no token).
Router log says repeatedly : user.notice OMR-VPS: Can't get token, try later
VPS syslog says encryption problem : 
Jan  3 06:25:11 vps ss-server[23814]: failed to handshake with 109.190.xxx.xxx: authentication error
Jan  3 06:42:57 vps ss-server[23814]: message repeated 71 times: [ failed to handshake with 109.190.xxx.xxx: authentication error]
etc 
## Steps to Reproduce the Problem

  1. ugrade to 53.2 router version
  2.upgrade of VPS
  

## Specifications

  - OpenMPTCProuter version: 53.2
  - OpenMPTCProuter VPS version: kernel 4.19.80-mptcp
  - OpenMPTCProuter platform: x86-64
[syslog.txt](https://github.com/Ysurac/openmptcprouter/files/4022151/syslog.txt)

suyuan168, 2020-01-06:
You should probably use the setup wizard to double check that your keys are filled in correctly.

jfh75, 2020-01-06:
Issue is same as "Can't get public IP address from shadowSocks #764"

closed 2020-01-06 by jfh75