Ysurac/openmptcprouter #786: Add Arch ar71xx

FollowBlackSheep, 2020-01-05:
Hey, i would like to use a TP-Link WR1043NDv4 for OpenMPTCPRouter is this possible at all?

i was trying to build a image by myself but i'm only getting the Error below:

Target ar71xx not found ! You have to configure and compile your kernel manually.

Ysurac, 2020-01-05:
CPU too slow with often not enough flash available.
Doc to compile it for not supported platform is here: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

suyuan168, 2020-01-06:
> Hey, i would like to use a TP-Link WR1043NDv4 for OpenMPTCPRouter is this possible at all?
> 
> i was trying to build a image by myself but i'm only getting the Error below:
> 
> Target ar71xx not found ! You have to configure and compile your kernel manually.

You can try linksys AC3200 to test. High performance.

FollowBlackSheep, 2020-01-06:
> linksys AC3200

Yeah but i don't have this Router so i wanted to use something i allready have.

closed 2020-05-18 by Ysurac