Ysurac/openmptcprouter #788: How to do a factory reset?

xabix, 2020-01-05:
Hello,

I have some instability with 0.53.2 and 1007 and I would like to reconfigure it from scratch. Is their an easy way to wipe to factory the config so I can start from scratch without having to rebuild the VM completely etc...

Merci
XabiX

Ysurac, 2020-01-05:
Upgrade without keeping the config.

closed 2020-01-05 by Ysurac

xabix, 2020-01-05:
@Ysurac

Merci

Is there a way to do it on the GUI?

Otherwise by the cmd, should I use the n option?

root@OpenMPTCProuter:~# sysupgrade 
Usage: /sbin/sysupgrade [<upgrade-option>...] <image file or URL>
       /sbin/sysupgrade [-q] [-i] [-c] [-u] [-o] [-k] <backup-command> <file>

upgrade-option:
        -f <config>  restore configuration from .tar.gz (file or url)
        -i           interactive mode
        -c           attempt to preserve all changed files in /etc/
        -o           attempt to preserve all changed files in /, except those
                     from packages but including changed confs.
        -u           skip from backup files that are equal to those in /rom
        -n           do not save configuration over reflash
        -p           do not attempt to restore the partition table after flash.
        -k           include in backup a list of current installed packages at
                     /etc/backup/installed_packages.txt
        -T | --test
                     Verify image and config .tar.gz but do not actually flash.
        -F | --force
                     Flash image even if image checks fail, this is dangerous!
        -q           less verbose
        -v           more verbose
        -h | --help  display this help

backup-command:
        -b | --create-backup <file>
                     create .tar.gz of files specified in sysupgrade.conf
                     then exit. Does not flash an image. If file is '-',
                     i.e. stdout, verbosity is set to 0 (i.e. quiet).
        -r | --restore-backup <file>
                     restore a .tar.gz created with sysupgrade -b
                     then exit. Does not flash an image. If file is '-',
                     the archive is read from stdin.
        -l | --list-backup
                     list the files that would be backed up when calling
                     sysupgrade -b. Does not create a backup file.

Ysurac, 2020-01-06:
You can do it via the gui by not checking the box save config when asked or via cli with the -n option.

xabix, 2020-03-28:
BTW using ext4 image for x86, I don't see any box "save config" in backup & restore. Is this in another place?
[image]

Ysurac, 2020-03-28:
To save the config "Generate archive"
If it's when you update, it's asked after clicking "Flash image"