Ysurac/openmptcprouter #790: Bug: VPN Tunnel down

rubaz100, 2020-01-06:
## Expected Behavior

Everything in the Network overview should have a green check (tick). But after a reboot (Router) I will get the message: "VPN Tunnel down". I need to restart the Glorytun VPN (UDP) everytime to see the green tick.

## Actual Behavior

But after a reboot (Router) I will get the message: "VPN Tunnel down". I need to restart the Glorytun VPN (UDP) everytime to see the green tick.

## Steps to Reproduce the Problem

  1. Reboot router (WRT3200ACM)
  2. Check the network overview 
  3. Restart Glorytun UDP

## Specifications

  - OpenMPTCProuter version: (0.53.3)
  - OpenMPTCProuter VPS version: (0.1007)
  - OpenMPTCProuter platform: (WRT3200ACM)

Ysurac, 2020-01-06:
I'm not able to reproduce the issue.

couili, 2020-01-08:
Bonjour, j'ai le même souci. Malgré ce bug d'affichage, l'agrégation fonctionne.

Ysurac, 2020-01-08:
I don't think it's not a bug in display, if it's displayed as down, then it's down.
You could check in Status->System log why omr-tracker thinks it's down and if glorytun UDP tell something.

couili, 2020-01-17:
Ce qui est curieux, c'est qu'en allant dans VPN > Glorytun puis en désactivant et réactivant le VPN, ça repasse au vert instantanément. J'avoue que je ne sais pas quelle partie des logs il vous faut.

couili, 2020-02-20:
Un ami vient de me dire qu'il pourrait y avoir des soucis entre la version 4.19 du kernel de linux, et tout ce qui concerne les VPN. Aujourd'hui, même après avoir tout redémarré, je n'arrive plus à avoir un VPN fonctionnel.

closed 2020-05-18 by Ysurac