Ysurac/openmptcprouter #793: Glorytun won't work on 0.53.3

dnwk, 2020-01-07:
## Expected Behavior
I could switch to Glorytun VPN in "advance setting" on setup wizard. 

## Actual Behavior

Admin API won't pull Glorytun VPN keys and settings. When manually switch to Glorytun, it still won't create Glorytun profile and pull the key.  Only option is DSVPN.

  - OpenMPTCProuter version: (Last is not a version)  0.53.3
  - OpenMPTCProuter VPS version: (Last is not a version) 0.1006
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) linksys wrt32x

Ysurac, 2020-01-07:
I'm not able to reproduce the issue.

dnwk, 2020-01-07:
Seems like related to config coming from previous version. When I reset my router to factory default, problem solved. But seem like something related to upgrading.