Ysurac/openmptcprouter #794: Can't aggregate 2 DSL connection

JorgeMoort89, 2020-01-07:
Hi Ysurac,
I'm having some issues bonding 2 ADSL (16 mbit/s download and 1 mbit/s upload each one) connection, I'm using the router remotely so sadly I can't upgrade him and try the latest version.

I have two issues:
1. When I start a download or omr-test-speed on the router I get very poor performance, omr-test-speed wan_numer is fine. I've also checked if the connection are MPTCP capable.
[image]

2. My log file is full of this route problem, but I can't figure out why, maybe a misconfiguration

Tue Jan  7 17:25:08 2020 user.notice omr-tracker: Can't create route to  via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan  7 17:25:09 2020 user.notice omr-tracker: Can't create route to  via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan  7 17:25:09 2020 user.notice omr-tracker: Can't create route to  via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan  7 17:25:10 2020 user.notice omr-tracker: Can't create route to  via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan  7 17:25:10 2020 user.notice omr-tracker: Can't create route to  via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan  7 17:25:11 2020 user.notice omr-tracker: Can't create route to  via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan  7 17:25:11 2020 user.notice omr-tracker: Can't create route to  via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan  7 17:25:12 2020 user.notice omr-tracker: Can't create route to  via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...

I've set omr-tracker to check only the gateway, otherwise the tunnel goes down after some seconds for a ping problem (Can't ping hosts IP)

My configuration: 

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/whatsapp.net/omr-cs2' '/whatsapp.com/omr-cs2' '/googleapis.com/omr-cs2' '/1e100.net/omr-cs2' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/steamcontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/gvt1.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/upload.youtube.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/whatsapp.net/omr-cs2' '/whatsapp.com/omr-cs2' '/googleapis.com/omr-cs2' '/1e100.net/omr-cs2' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/steamcontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/gvt1.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/upload.youtube.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/whatsapp.net/omr-cs2' '/whatsapp.com/omr-cs2' '/googleapis.com/omr-cs2' '/1e100.net/omr-cs2' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/steamcontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/gvt1.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/upload.youtube.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/whatsapp.net/omr-cs2' '/whatsapp.com/omr-cs2' '/googleapis.com/omr-cs2' '/1e100.net/omr-cs2' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/steamcontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/gvt1.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/upload.youtube.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/whatsapp.net/omr-cs2' '/whatsapp.com/omr-cs2' '/googleapis.com/omr-cs2' '/1e100.net/omr-cs2' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/steamcontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/gvt1.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/upload.youtube.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/fbcdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/v.redd.it/omr-cs4' '/ttvnw.net/omr-cs4' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/google.com/omr-cs2' '/fbcdn.n [... cut ...]
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.9962
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-01-07:
Seems that only pinging GW on this release doesn't work. I thinks it's solved in newer release.
Are you sure that each GW answer to ping because only wan1 seems to be detected as up ?

JorgeMoort89, 2020-01-07:
Yes, I've noticed that on the status page WAN3 is showed as DOWN and I don't know why, the gateway is pingable

JorgeMoort89, 2020-01-07:
Is the "Seems that only pinging GW on this release doesn't work. I thinks it's solved in newer release." resolved in the newer version?

Ysurac, 2020-01-07:
Maybe

JorgeMoort89, 2020-01-07:
I've setted again the router to use Glorytun TCP + SS, I've notice that I have the download stucked at 8 mbit/s aggregate (it have to be 40 mbit/s) but when I get 8 mbit/s in download I get the upload saturated, how can I fix that? Fixing the MTU?

Ysurac, 2020-01-07:
You have a really old release, I don't remember default settings of it...
You can use SQM and lower value while it's not ok.

JorgeMoort89, 2020-01-07:
SQM helped a little bit, i've reached 12 mbit/s in download fixing thee upload a 700 kbit/s, how can I optimize it a little bit? I'll try the new version soon

Ysurac, 2020-01-07:
I would say to try other MPTCP scheduler and TCP congestion crontrol, but it's limited on your version. You can try to fix lower than 1500 MTU, I don't think there is auto MTU discovery on this release.

JorgeMoort89, 2020-01-08:
Tried a lot of scheduler and TCP congestion control, now I'm trying to upgrade to the latest version with a fresh installation, I'm checking if the gateway check problem is solved.

JorgeMoort89, 2020-01-08:
On 0.52.3 seems resolved the gateway problem, but in the diagnostic when I try to ping some host i receive this error "PermissionError: Permission denied". How on this version tune OMR to use all the download?

JorgeMoort89, 2020-01-08:
Another problem, I've disabled on omr-tracker the check for the connection down but if I block ICMP on my router OMR turn off the connection and give the error "eth1 switched off because gateway down".

JorgeMoort89, 2020-01-08:
@Ysurac how can I fix manually the route to gateway problem? Will be a huge help, I have to go to the router location to upgrade it. Thanks

Ysurac, 2020-01-08:
To work correctly OpenMPTCProuter need omr-tracker to work. If type is none it will ping only gateway, else it ping external website (can be http test or dns also).
If type is none and gateway doesn't answer to ping, then the interface is set to down (I have no way to know the status). So why do you block ICMP ? ICMP should always answer.

JorgeMoort89, 2020-01-08:
The gateway isa reachable, when I set the tracker to none and also disable that I hve the log full of 
Tue Jan 7 17:25:08 2020 user.notice omr-tracker: Can't create route to via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan 7 17:25:09 2020 user.notice omr-tracker: Can't create route to via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan 7 17:25:09 2020 user.notice omr-tracker: Can't create route to via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan 7 17:25:10 2020 user.notice omr-tracker: Can't create route to via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan 7 17:25:10 2020 user.notice omr-tracker: Can't create route to via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan 7 17:25:11 2020 user.notice omr-tracker: Can't create route to via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting...
Tue Jan 7 17:25:11 2020 user.notice omr-tracker: Can't create route to via 192.168.1.1 dev eth3 src 192.168.1.83. waiting...
Tue Jan 7 17:25:12 2020 user.notice omr-tracker: Can't create route to via xxx.xxx.xxx.xxx dev eth1 src xxx.xxx.xxx.xxx. waiting... 

If you notice the destination host is missing in the log becouse I've deleted it on OMR

Ysurac, 2020-01-08:
I'm not able to reproduce this issue with latest release, when type is set to none it shouldn't try to create a route to something that doesn't exist.

JorgeMoort89, 2020-01-08:
Can direct upgrade from 0.49 to 0.52.3 fail a corrupt? Maybe I can try upgrading reotely via teamviewer

Ysurac, 2020-01-08:
Yes, this can. It's an old release, and VPS script upgrade is also required.

JorgeMoort89, 2020-01-14:
I've found the problem, if I remove the route "default via 10.255.255.1 dev tun0" I can get the aggregation, If I add it with "ip route add default via 10.255.255.1 dev tun0" It turn slow, how can I fix it?

Ysurac, 2020-01-14:
This route use Glorytun TCP VPN, it's used only for anything that is not TCP. Shadowsocks is used for TCP.
How you test speed to say it's slow ?

JorgeMoort89, 2020-01-14:
omr-test-speed, if I add that route all the traffic pass the VPN and it slows a lot, how can I add a route only for UDP traffic and natted port?

Ysurac, 2020-01-14:
If shadowsocks is enabled (it's the default) then all TCP traffic use shadowsocks.

JorgeMoort89, 2020-01-14:
I've initially disabled ss, after that I've re-enabled it via uci with the command "openmptcprouter.settings.shadowsocks_disable='0'", I think that in this case it is sending all the traffic on the VPN, how can I change that? Are there some rule to route all the TCP traffic con SS?

closed 2020-05-18 by Ysurac