Ysurac/openmptcprouter #797: Netword overview cannot be displayed

8252610, 2020-01-15:
Install the latest router after I initialize the disk (0.53.3)
Install the latest VPS after I initialize the disk（1008）
Netword overview cannot be displayed
But vps1007 is all right
[image]
[image]
[image]
  - OpenMPTCProuter version: (0.53.3)
  - OpenMPTCProuter VPS version: (1008)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-01-15:
I will fix the status page but it's not working because you have a config problem. I think it's a problem with the interfaces configurations.

8252610, 2020-01-16:
think you very much

closed 2020-05-18 by Ysurac