Ysurac/openmptcprouter #798: VPS Installation with Ubuntu 18.04 Freezes on Restart

legolas108, 2020-01-15:
## Expected Behavior

After installing VPS side with Ubuntu 18.04 (Vultr) system should reboot to the MPTCP kernel.

## Actual Behavior

The boot process freezes.

## Steps to Reproduce the Problem

Create new VPS instance on Vultr with Ubuntu 18.04. Run OpenMPTCProuter installation script and reboot as requested.

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0
  - OpenMPTCProuter VPS version: Ubuntu 18.04
  - OpenMPTCProuter platform: x86_64

## Comments

After installing with Debian 10 on Vultr and successfully rebooting and configuring USB tethered smartphones by adding their `usbN` interfaces system works very nicely and is very stable. Great tool - thanks so much!

Ysurac, 2020-01-15:
I'm not able to reproduce the issue on Vultr with Ubuntu 18.04

closed 2020-01-15 by Ysurac

legolas108, 2020-01-15:
Thanks for really quick response. Just tried it again. $10 Vultr instance with Ubuntu 18.04, only

`apt update && apt full-upgrade -y`

and your installation command

`wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh`

Logged out, restarted from Vultr interface and you can see where it hangs in the picture:

[image]

Ysurac, 2020-01-16:
Even if Vultr console stop here, the server boot without problems. You can ssh to the server port 65222 and check ```/var/log/kern.log```.

legolas108, 2020-01-16:
Ran the same setup on a Linode with virtually same VPS specs and there it finished flawlessly and one can login from the Console window, too. That's important for me as I tend to lock myself out from SSH access every now and then ;-)

Thanks so much for your great tool and support!