Ysurac/openmptcprouter #799: Décodeur TV orange down après installation

GS-m-007, 2020-01-19:
Bonjour,

Tout d'abord, un grand merci et toute mes félicitations pour cette solution au top !! 
J’ai fais à mon tour une installation selon la configuration suivante :
– box 4G Bouygues
– box internet Orange
– Raspberry Pi 3
– vps chez Onesolutions ( j’ai essayé avec VPS Firstheberg mais blocage de Netflix !! :/)

Mon soucis est maintenant de faire fonctionner mon décodeur tv Orange en mode « Agrégé » ??
Comment faire? Une solution pour palier au soucis ??

Merci d’avance pour le retour.
Bonne journée

Ysurac, 2020-01-19:
Orange doit limité le décodeur a ses propres IPs, donc je pense qu'il n'est pas possible pour le décodeur TV d'Orange d'utiliser l'agrégation car l'IP du VPS est utilisé dans ce cas.
Sinon hors agrégation il doit être toujours possible de le relier à la box Orange ou via un vlan (mais le décodeur ne profitera pas de l'agrégation).

GS-m-007, 2020-01-19:
Ok donc pas possible dans l état de rentrer le décodeur tv dans la configuration. Après cela ne me gêne pas spécialement. Vous parlez d une autre solution: pouvez-vous m en dire plus si vous maîtrisez le sujet et le donner des indications pour réaliser le truc? (Mise en place d un vlan entre box et décodeur)

Merci d avance.

Ysurac, 2020-01-19:
Je n'ai pas de box Orange donc je ne peux pas en dire plus, mais le soucis est le même quelque soit le routeur utilisé, je pense que l'on peut trouver de la doc sur le sujet sur internet.

GS-m-007, 2020-01-19:
Ok je vais voir ce que je trouve comme doc sur la mise en place d un vlan entre box et décodeur Orange. Si je trouve quelque chose et que cela fonctionne je mettrai en ligne. Si quelqu’un a des informations je suis preneur...
En tout cas je vous remercie pour votre retour. Bonne soirée.

Randdrick, 2020-01-22:
Bonjour,
C'est normal que le décodeur orange soit down après l'installation puisque sa "configuration" se fait automatiquement par la box orange. 
Plusieurs ports sont ouverts au niveau du pare-feu de cette dernière, mais je pense qu'ils n'apparaissent pas tous dans l'interface.
C'est la raison pour laquelle, le DHCP doit être actif sur la box Orange pour que cette dernière délivre une adresse IP au décodeur TV.
Afin que cela fonctionne sans problème, j'ai procédé ainsi : 
- Au niveau DHCP, j'ai indiqué que mon routeur est le seul serveur DHCP du réseau. Le DHCP est in use pour uniquement l'interface LAN désactivé pour les interfaces WAN.
- J'ai interdit la fourniture d'une adresse IP pour l'adresse MAC du boitier TV Orange.

Au niveau du décodeur Orange, j'ai : 
- Débranché le câble réseau reliant mon décodeur au routeur.
- Redémarré mon boitier TV pour que ma box lui donne bien son adresse IP et le configure automatiquement.
- Rebranché le câble réseau reliant mon décodeur au routeur.

xillibit, 2020-01-28:
>     * J'ai interdit la fourniture d'une adresse IP pour l'adresse MAC du boitier TV Orange.

Bonsoir,

Je ne trouve pas comment faire ça dans openmptcprouter, pouvez-vous m'indiquer ou ça se trouve ?

GS-m-007, 2020-01-29:
Bonjour,

merci pour votre réponse ...
comment réaliser cela dans la configuration d'openwrt ??
merci d'avance pour votre retour.

closed 2020-09-14 by github-actions[bot]