Ysurac/openmptcprouter #801: VOIP registry timeout after a moment

siliteofr, 2020-01-20:
## Expected Behavior

Hi,

With OMTCP telephony work's great (asterisk server on an hosted server). But Some phone start to stop register after 4-5 hours (but before they works great). We have to reboot OMTCP Raspi router : if i don't,  all phones stop to register before the end of the day (And no working telephony anymore).

The problem can be solved by:
- restart the phone that can't register
- restart the OpenMTCP.

No problem if i use one of xDSL connection. Problem occurs on different site, wich all have : 
- an OMTCP Router using a different VPS.
- a different voip server
- a different brand of phone

It's like a SIP ALG problem. But i think it's disabled on OMTCP?

## Actual Behavior

## Steps to Reproduce the Problem

  1. Wait 2-3 hours with a phone using Centrex

## Specifications

  - OpenMPTCProuter version: 0.53 or 0.52
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: RPi3

Ysurac, 2020-01-20:
SIP ALG is not supported at all, so no problem here.
Can you try with another VPN ? By default it's Glorytun TCP

closed 2020-09-14 by github-actions[bot]

Malaga82, 2020-10-07:
Hello, i have SIP problem too.
v0.55.3 r0+14198-da98603597 for router
Version 0.1017 5.4.52-mptcp for vps
with siemens A510 IP i can't get registration from FTTC provider that give me the number.
the protocol used must be UDP, i can't use TCP.
Sometimes it register but if i call the number it doesn't hook the call, as soon as i change the gateway to one of my wans, it register and if call the number it hooks the call fine.
It will be glad if this issue can be fixed, since all fttc provider gives voip number ( at least here ).
Thanks