Ysurac/openmptcprouter #803: No more wifi with 0.53

yann20, 2020-01-22:
## Expected Behavior

I am not an expert under linux and since my passage with version 0.53 I have a little problem

My wifi hotspot does not wake up on its own.
I have to connect each time by wire to my pi then go to "network" then "wireless" then restart on "radio0" genericmac80211

With version 0.52 this was done automatically.

Can you help me?

## Actual Behavior

## Steps to Reproduce the Problem

  1. Con,figure wifi on
  2. reboot 
  

## Specifications

  - OpenMPTCProuter 0.53
  - OpenMPTCProuter VPS version: Bedian 10 buster 64
  - OpenMPTCProuter platform: RPI3 B+

quellhorst, 2020-03-04:
I had this same issues on a RPI 3 B+ and did a quick fix by adding the following to /etc/rc.local 

wifi down radio0
wifi up radio0

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. This should be fixed in release.
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac