Ysurac/openmptcprouter #804: macvlan on VLAN interface?

jayanta525, 2020-01-24:
## Expected Behavior
Tagged VLAN 1866 and 1867. DHCP enabled on both network.
in /etc/config/network macvlan wan1 master interface is set as eth0.1866. 
and wan2 master interface is set as eth0.1866,
and wan3 master interface is set as eth0.1867.

## Actual Behavior
Interface fails to obtain IP, setting static IP also fails, no gateway ping on all the three interfaces.

## Steps to Reproduce the Problem

  1. Create two networks.
  2. Tag with VLAN ID
  3. Configure in OpenMPTCPRouter

## Specifications

  - OpenMPTCProuter version: 0.53
  - OpenMPTCProuter VPS version: 0.1005
  - OpenMPTCProuter platform: RPI3

jayanta525, 2020-01-24:
Works fine on OpenMPTCProuter version 0.52

Ysurac, 2020-01-24:
MacVLAN over VLAN ? For me this never worked because it's vlan over vlan.

jayanta525, 2020-01-24:
I managed to make it work on 0.52 with
[image]

With the UI changes in 0.53. The same configuration fails to work. Anyways, its working great so far. Thanks.

Ysurac, 2020-01-24:
ok, I will add a custom fields to MPTCP interface for next release.

jayanta525, 2020-01-25:
Supporting VLAN in OMR will benefit for boards with single Ethernet port. With VLAN the single interface can be extended into multiple REAL Ethernet ports with a managed switch (or OpenWrt based router).

closed 2020-01-30 by Ysurac