Ysurac/openmptcprouter #805: All Google traffic does NOT go through VPN

max4186, 2020-01-27:
For some reason, ALL traffic for Google does not go through the VPN. Especially Google Drive. I have no bypass setup for Google whatsoever, is there something else that could cause it?

Ysurac, 2020-01-27:
Check that your computer doesn't get an IPv6 from another modem/box.

max4186, 2020-01-27:
Unfortunately, no IPv6 enabled :(

Ysurac, 2020-01-27:
How do you know that VPN is not used ?

max4186, 2020-01-27:
2MB/s internet vs 15MB/s via bonded.

Ysurac, 2020-01-27:
If you use google chrome I think that Google Drive use QUIC that is based on UDP and use the VPN (by default Glorytun TCP), else it should use HTTP/2 based on TCP that use shadowsocks and should be faster for our case.

max4186, 2020-01-27:
Ah.... that explains it. I have the UDP VPN turned off completely as it causes latency issues for me - let me see if I can disable QUIC and test again.

Ysurac, 2020-01-27:
You can disable it in Chrome or block port 443 and 80 in UDP on the firewall of OpenMPTCProuter.

max4186, 2020-01-28:
This can be closed, it was indeed QUIC.

closed 2020-01-29 by Ysurac