Ysurac/openmptcprouter #806: Glorytun UDP not working

SriramScorp, 2020-01-27:
Since ISP is blocking MPTCP, disabled shadowsocks and changed glorytun protocol to UDP.

## Expected Behavior
Tunnel should be up and use Glorytun UDP for all traffic.

## Actual Behavior
1. Status page shows 'VPN tunnel Down'.
2. In router, glorytun's localip and remoteip remains the same. Doesn't change to '10.255.254.2' and '10.255.254.1' for udp.
3. In vps, parameters are not updated in /etc/shorewall/params.vpn

## Steps to Reproduce the Problem
  1. Services -> Shadowsocks-libev -> Remote Servers -> Disable
  2. VPN -> Glorytun -> proto -> udp

## Specifications
  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: RPI3

```
# uci show openmptcprouter | grep mtu
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan2.mtu='1410'
openmptcprouter.wan3.mtu='1354'
```
While the low MTU is related to #485 , I don't even have the tunnel UP and glorytun path is always 'DEGRADED'. Tried manually setting tx/rx as mentioned in [https://github.com/angt/glorytun/issues/40](url) but didn't help.
```
# glorytun-udp show dev tun0
client tun0:
  pid:    1551
  bind:   0.0.0.0 port 5000
  peer:   *REDACTED* port 65001
  mtu:    1280
  cipher: chacha20poly1305
```
```
# glorytun-udp path dev tun0
path UP
  status:   DEGRADED
  bind:     192.168.8.113 port 5000
  public:   - port 0
  peer:     *REDACTED* port 65001
  mtu:      1302 bytes
  rtt:      0.000 ms
  rttvar:   0.000 ms
  rate tx:  125000000 bytes/sec
  rate rx:  125000000 bytes/sec
  total tx: 488 packets
  total rx: 0 packets
path UP
  status:   DEGRADED
  bind:     192.168.14.100 port 5000
  public:   - port 0
  peer:     *REDACTED* port 65001
  mtu:      1302 bytes
  rtt:      0.000 ms
  rttvar:   0.000 ms
  rate tx:  125000000 bytes/sec
  rate rx:  125000000 bytes/sec
  total tx: 486 packets
  total rx: 0 packets
```
What can I do to get Glorytun UDP working?

Ysurac, 2020-01-27:
Use the wizard to change the VPN

SriramScorp, 2020-01-29:
Glorytun VPN is enabled and the service is running. It can be confirmed from ps.
```
# ps | grep [g]lorytun
 1551 root      1128 S    /usr/sbin/glorytun-udp bind 0.0.0.0 to *REDACTED* 65001 dev tun0 keyfile /tmp/glorytun-udp-vpn
```
The same process ID is also reported by `glorytun-udp show dev tun0`. No other VPN is enabled or running.

SriramScorp, 2020-01-29:
Well, disabling Shadowsocks wasn't effected and was showing as unsaved changes. After saving the changes, tunnel is UP. But, does that mean Glorytun-UDP cannot be used when Shadowsocks is up? Or can I just remove the redir rules without completely disabling Shadowsocks to get Glorytun-UDP working?

Ysurac, 2020-01-29:
You use an old release of OpenMPTCProuter, I don't remember the issues.

closed 2020-08-12 by Ysurac