Ysurac/openmptcprouter #807: Installing updates via opkg

slitsevych, 2020-01-27:
##
It's rather a simple question than an issue:
Is it safe to install any "opkg" updates of packages installed by default? 
I can see that OpenWrt released a new stable version 19.07.0 and "opkg list-upgradable" shows lots of new versions but I'm not sure if it's safe to simply install these updates. 
What do you suggest @Ysurac  ?

Thank you in advance.

## Specifications

  - OpenMPTCProuter version: 0.53.5
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: x86-64 (64-bit) ext4 image (EFI) installed on KVM machine

Ysurac, 2020-01-27:
For now it's not recommended and not tested.

slitsevych, 2020-01-28:
Thank you @Ysurac, got it! 

Could you please also specify, have you ever tested or tried implementing any Zabbix monitoring of OpenWRT environment, for example using templates and user parameters like described here:
https://wiki.commotionwireless.net/doku.php/general_resources/documentation/monitoring/zabbix

Thank you in advance again!

Ysurac, 2020-01-29:
no

closed 2020-01-29 by slitsevych