Ysurac/openmptcprouter #808: internet cutoff

CoSysopK, 2020-01-28:
## Actual Behavior

**When there are these logs :**
daemon.err /usr/bin/ss-redir[10359]: remote recv: Operation timed out
daemon.err /usr/bin/ss-redir[10360]: remote recv: Operation timed out
daemon.err /usr/bin/ss-redir[10361]: remote recv: Operation timed out
daemon.err /usr/bin/ss-redir[10362]: remote recv: Operation timed out
**And these logs :** 
daemon.err glorytun[4621]: read: Operation timed out

I no longer have internet for a few seconds.
This is a problem when I stream because this problem causes a cut.

## Steps to Reproduce the Problem

  1. Install image for Raspberry PI 3/3B+ (64-bit)
  2. Install VPS Ubuntu Server 18.04 x86_64
  3. Configure the router with the wan1 static and wan2 static
  4. Add the IP address VPS and, all the informations (shadowsocks key, glorytun key, MLVPN
password and OpenMPTCProuter VPS Admin key) retrived on the VPS
  5. Two keys 4G on usb PI 3B

## Specifications

  - OpenMPTCProuter version: Version 0.52.3
  - OpenMPTCProuter VPS version: Version 0.1001
  - OpenMPTCProuter platform: RPI3B+ 64 bits

Ysurac, 2020-01-29:
If it's a problem with a wan, you can modify omr-tracker timeout.
If it's a problem with the VPS, choose another one.

CoSysopK, 2020-01-29:
Okay. First, I will do configuration tests about omr-tracker.
In a second step, I will change VPS if necessary.

CoSysopK, 2020-01-29:
This type of log caused an interruption in my connections for a few seconds.

Wed Jan 29 17:10:55 2020 daemon.err glorytun[4714]: read: Operation timed out
Wed Jan 29 17:10:55 2020 daemon.info glorytun[4714]: STOPPED tun0
Wed Jan 29 17:10:56 2020 daemon.err glorytun[4714]: 35.180.50.248.65001: connected
Wed Jan 29 17:10:56 2020 daemon.info glorytun[4714]: STARTED tun0

What should I do ?

Ysurac, 2020-01-29:
You can try another VPN. This error should interrupt only what is not TCP traffic, it's a timeout because packets can't reach the VPS.

CoSysopK, 2020-01-29:
Currently I am at Amazon.
From experience, which VPN would you recommend?
(to have an additional opinion during my choice)

CoSysopK, 2020-01-31:
I put these settings and for the moment it's stable.
No disconnection from the stream
No "Attempt to connect" popup, in network for 10 seconds.

[image]
[image]
[image]

closed 2020-02-05 by CoSysopK