Ysurac/openmptcprouter #809: Wifi DHCP on iPhone

ebuyfr, 2020-01-28:
## Expected Behavior

I have a problem with my new version v0.53 installed on a Pi4: DHCP not working with my iPhones (only the IPV6 works).
My configuration is a Freebox Delta and a 4G Netgear Router.
During the first installation everything worked fine, then I decided to add Pi-Hole, since everything went wrong.
I reinstalled everything OpenMPTCPRouter from scrach several times and yet I still end up with this error that I cannot correct: I clearly see my iPhones taking IPV4 addresses successively but they do not keep them ...
I blew the DHCP IPV6 on the Freebox yet etc ...
Is this kind of problem known?
Thank you in advance.

## Actual Behavior
All products in Wifi connection are ok. Only my iPhones no good.

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.53.3 r0+11546-a74095c68c / LuCI Master git-19.354.24352-2096d67
  - OpenMPTCProuter VPS version: VPS 0.1006
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

Ysurac, 2020-01-29:
It's a know issue of iOS DHCP client. No idea if this can be solved.

ebuyfr, 2020-01-29:
Argh pourtant ça fonctionnait avant que je ne bricole...

ebuyfr, 2020-01-29:
N'y a t'il pas une réponse ici [https://forum.netgate.com/topic/84163/domain-mydomain-local-issue-dns-dhcp-mdns-apple](url)

Ysurac, 2020-01-29:
J'utilise pas .local pour OpenMPTCProuter mais .lan, donc je pense pas.

Ysurac, 2020-01-30:
D'après la doc OpenWrt ( https://openwrt.org/docs/guide-user/base-system/dhcp.dnsmasq ) pour des soucis similaire on peut régler le soucis via SSH sur le router en tapant ça : 
```
uci add_list dhcp.lan.dhcp_option='252,"\n"'
uci commit dhcp
service dnsmasq restart
```

ebuyfr, 2020-01-30:
Un grand Merci , je vais tester ce we. ;)

ebuyfr, 2020-01-30:
Tellement pressé j'ai testé ce soir. Rien de mieux. Je vois bien le paramétrage dans Interface/Lan0/Servuer DHCP/Options Avancées : Options DHCP : 252,"\n"
Mais toujours pas de DHCP pour mes iPhones...
Je vais retenter ce we.

dabuide, 2020-02-11:
Visiblement le souci est lié à Apple complètement, j'ai un souci similaire sur le Mac et même en lui attribuant une adresse IP statique il me dit que l'IP est déjà utilisé alors que c'est pas le cas. Il n'y a que les équipements Apple qui buguent !

ebuyfr, 2020-02-11:
Salut, du coup je rebondis, le probleme se passe sur les Apple oui mais de mon côté c’est un paramétrage chez Free qui a changé le comportement de mon installation. J’ai depuis mis le routeur 4g en coupure derrière un routeur et je n’ai plus le problème. 
Ma config : 
1 Adsl Free 
1 routeur 4g Netgear (SIM Free)
1 raspberry 3 connecté
1 switch netgear reliant tout le monde. 
Depuis j’ai ajouté un tp-link entre le routeur 4g et le switch. 
J’ai aussi passé le routeur 4g en mode pont. 
Mon problème est résolu avec les appareils Apple. 
Me reste à faire fonctionner Pi-Hole mais c’est un autre sujet...

dabuide, 2020-02-11:
Je vois, je pense que mon problème viens aussi de OpenWRT étant donné que le seul point de contact entre tout mes équipements est la liaison wifi avec le router au milieu de la maison. Le DHCP fonctionne dans les logs mais il ne prend aucune adresse est validé. Un travail de profondeur à faire. Pour la parti pi-hole je peux donner un coup de main j'en ai deux à la maison

ebuyfr, 2020-02-11:
Merci pour l’offre, en local ça fonctionne presque tout seul mais j’ai tenté l’intégration sur le VPS, il est en place mais rien. 
Concernant le dhcp moi je vois que les adresses sont prises les unes après les autres par les iPhones, en farfouillant dans les paramètres IPV6 de la connexion Wifi, puis en revenant au menu d’avant on voit apparaître une adresse IP, mais elle ne reste pas. Les baux sont pris successivement les uns après les autres et relâchés dans la foulée. Grâce à ma Freebox je le vois très clairement en monitorant mes iphones ou tablettes.

Ysurac, 2020-02-11:
Pi-Hole sur le VPS doit être via le script prévu dans la doc. Ensuite une nouvelle validation du wizard dans Système->OpenMPTCProuter peut être nécessaire.

Pour le DHCP, je n'ai aucune idée du soucis, ma configuration est celle par défaut d'OpenWRT qui n'a rien de spécial. Certains appareil Apple refusent toutes les IPs proposés, il y a peut être une vérification via ARP de la disponibilité des IPs, mais je ne vois pas pourquoi il y aurait une réponse pour une IP non utilisée. Visiblement il y avait le même soucis sur des routeurs Asus utilisant aussi DNSMasq mais ils semblent avoir régler le problème via une mise à jour du firmware, mais je ne sais pas comment...

dabuide, 2020-02-11:
C'est justement le souci avec un routeur Asus je pense que je vais downgrader le firmware visiblement les souci sont apparus depuis que j'ai mis à jour le routeur.

ebuyfr, 2020-02-11:
Par contre en IPV6 tout va bien, il n’y a plantage que pour IPV4. Si j’avais su utiliser ipv6 j’aurai délaissé ipv4 mais je n’y ai rien compris. 
Pour info j’ai bien utilisé le scrip prévu. Mais je n’ai pas relancé de validation derrière. 
En tout cas ça fonctionne très bien au niveau agrégation ;)

dabuide, 2020-02-11:
pas mieux :
`
Tue Feb 11 20:19:35 2020 daemon.info dnsmasq-dhcp[32558]: DHCPDISCOVER(eth0) c0:a5:3e:99:1b:af
Tue Feb 11 20:19:35 2020 daemon.info dnsmasq-dhcp[32558]: DHCPOFFER(eth0) 192.168.100.227 c0:a5:3e:99:1b:af
Tue Feb 11 20:19:37 2020 daemon.info dnsmasq-dhcp[32558]: DHCPREQUEST(eth0) 192.168.100.227 c0:a5:3e:99:1b:af
Tue Feb 11 20:19:37 2020 daemon.info dnsmasq-dhcp[32558]: DHCPACK(eth0) 192.168.100.227 c0:a5:3e:99:1b:af AppleWahdeDavid
Tue Feb 11 20:33:52 2020 daemon.info dnsmasq-dhcp[32558]: DHCPREQUEST(eth0) 192.168.100.227 c0:a5:3e:99:1b:af
Tue Feb 11 20:33:52 2020 daemon.info dnsmasq-dhcp[32558]: DHCPACK(eth0) 192.168.100.227 c0:a5:3e:99:1b:af AppleWahdeDavid
Tue Feb 11 20:33:53 2020 daemon.info dnsmasq-dhcp[32558]: DHCPDECLINE(eth0) 192.168.100.227 c0:a5:3e:99:1b:af`

dabuide, 2020-02-12:
visiblement c'est bien dnsmasq qui pose problème, du coup quel est l'interface que tu utilises en DHCP @ebuyfr ?

ebuyfr, 2020-02-12:
J’ai remis Mptcprouter comme dhcp. Comme expliqué plus haut c’est le routeur 4G qui empêchait le dhcp vers les périphériques Apple (hors Apple TV je crois). Mais cela avait fonctionné avant pendant plusieurs jours. Je pense à une mise à jour côté Free.

dabuide, 2020-02-12:
Je viens de remettre un routeur 4G Netgear récemment je vais regarder de ce côté-ci alors

ebuyfr, 2020-02-12:
J’ai du le passer en bridge pour obtenir un fonctionnement normal. Bon courage.

dabuide, 2020-02-12:
Du coup avec le mode bridge tu peux utiliser la connexion internet 4G vers l'MPTCP car je viens de le faire j'ai juste le routeur qui a pris une adresse DHCP du MPTCP mais plus accessible

ebuyfr, 2020-02-12:
C’est presque ça : j’ai ajouté un routeur Tp-link entre les 2 et là ça reprend de l’IPV4.

dabuide, 2020-02-12:
Donc si je comprend bien tu as la BOX FREE + routeur 4G pour MPTCP et entre les deux tu as rajouter un routeur TP-link ?

ebuyfr, 2020-02-12:
Je crois l’avoir déjà expliqué plus haut. Adsl Free en sortie un switch Netgear où est pluggé le MpTcpRouter mes périphériques et le Tp-link. 
Auparavant, sans le Tp-link et le routeur 4g en mode routeur ça fonctionnait. Puis j’ai voulu insérer un Pi-Hole et je ne me suis pas aperçu du fonctionnement Dhcp car c’était le Pi-Hole qui gérait et surtout j’ai as presque tous les périphériques en IP fixe. 
Donc je n’arrive pas à déterminer à quel moment ça a commencé à partir en cacahuète...
J’ai tout refait 4 ou 5 fois et je n’ai jamais réussi à refaire fonctionner le Bouzin, c’est pour cette raison que je suis venu ici. 
Moi qui voulait éliminer mon Tp-link ....
Aussi j’ai gardé le Dhcp de la Freebox pour adresser le Player en Wifi (Actif à chaque reboot). 
Voilà je crois que j’ai tout dit 😉

dabuide, 2020-02-13:
Bon après plusieurs heures de test, de reconfiguration, j'ai trouvé une alternative, j'ai brancher un routeur qui prend pour WAN le réseau MPTCP, et qui à son propre DHCP profite de la bande passante du MPTCP et lui se connecte sans problème en Wifi avec tous les équipements

closed 2020-09-14 by github-actions[bot]