Ysurac/openmptcprouter #810: feature request: add  bridging support

benoitc, 2020-01-29:
while we can create a DMZ by forwarding all port to a machine it would be interresting to have a way to put the router in bridge mode. This would allow to create "simple" bonding boxes  and use other hardware behind for the final routing and firewalling.

Ysurac, 2020-01-29:
I don't think it's possible with OpenMPTCProuter for now (I don't know if this would work with shadowsocks), but you are free to try networks settings.

user747, 2020-02-05:
containers might be a possible solution for this?  i think when you add more layers of abstraction, complexity goes up and performance possibly takes a hit. maybe running 1 container with 4 WAN, you also then possibly double or triple the number of parameters you are managing (like static routes in the container and on the host, firewall rules)

closed 2020-09-14 by github-actions[bot]