Ysurac/openmptcprouter #813: Intermittent HTTPS Connection Failures

legolas108, 2020-01-31:
## Expected Behavior

Browser pages load normally via HTTPS, other HTTPS traffic e.g. from command line finishes normally.

## Actual Behavior

Intermittently pages do not load with a PR_END_OF_FILE_ERROR. Java programs display "javax.net.ssl.SSLHandshakeException: Remote host closed connection during handshake" errors, "Caused by: java.io.EOFException: SSL peer shut down incorrectly". Git clones fail like "fatal: unable to access 'https://gitlab.linphone.org/BC/public/linphone-android.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated."

Retrying (sometimes several times) usually completes the intended action successfully, in rare cases OMR requires a restart.

## Steps to Reproduce the Problem

Cannot be reliably reproduced, happens randomly. Currently running a single client only with manual static IP.

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: x86_64
  - 6 smartphones tethered via USB
  - left the predefined devices wan1 and wan2 un-configured as I couldn't delete them

## Comments

Not sure if it has to do with this, but also intermittently "Multipath seems to be blocked on the connection" is displayed as a warning on some of the connections and this warning usually disappears after a short while by itself.

Another thing that should probably not be like that: under Status / Routes some of the target addresses display "0.0.0.0%teql0/0" or "192.168.42.0%teql0/24" instead of "0.0.0.0/0" or "192.168.42.0/24" like for corresponding others.

Haven't experimented, yet, but would changing some of the advanced network settings help?

Ysurac, 2020-02-02:
Can you test with only one ss_redir enabled in shadowsocks ?

legolas108, 2020-02-04:
Thanks! Disabled all ss_redir instances except the top one (ss_redir.hi). After a few hours the HTTPS connection failures have unfortunately occurred already again a few times.

Ysurac, 2020-02-04:
Can you check Status->System log when this happen ? You may have a ss_redir error.
This happen on any site ? When this happen can you try with another browser ?

legolas108, 2020-02-04:
The error can happen on any site, even www.google.com or when following search results. There are quite a few ss_redir errors like

    Tue Feb  4 06:52:11 2020 daemon.err /usr/bin/ss-redir[23884]: remote recv: Operation timed out

or

    Tue Feb  4 06:52:15 2020 daemon.err /usr/bin/ss-local[23883]: getpeername: Socket not connected

These are basically the only errors in the system log. (A few others I can explain and seem not related.)

legolas108, 2020-02-04:
OK, after several hours without issues now there happened another connection failure (PR_END_OF_FILE_ERROR) when opening new tab in Firefox (www.google.com):

    Tue Feb  4 11:26:27 2020 daemon.err /usr/bin/ss-redir[23884]: remote recv: Operation timed out

An immediate reload was successful, and loading the same page in Chrome right after was also no problem. It feels much improved with only one ss_redir instance. Is there a disadvantage compared to the default 4 instances?

Ysurac, 2020-02-04:
This can be slower because Shadowsocks is single threaded, so the only way to use multiple CPU core is loading multiple ss-redir.
I will try to find a good way to fix this issue.

legolas108, 2020-02-04:
I see. Just ran a few speed tests to compare. Though it's highly variable speed we get with the 4G connections one ss_redir instance seems to be slightly slower indeed (like 15 vs 13 Mbps).

Thanks so much for your tremendous work on this!

legolas108, 2020-03-04:
Found a hint in [this issue for shadowsocks-libev](https://github.com/shadowsocks/shadowsocks-libev/issues/2094#issuecomment-403053323). The author solved "random lost connection attempts" by specifying `LimitNOFILE=99999` and `LimitNPROC=99999` in the systemd config file for the ss-server service because by default only 4000 open files were allowed for it.

HTH :-)

Ysurac, 2020-05-18:
This will be solved in 0.55. If not, you are free to reopen this issue.

closed 2020-05-18 by Ysurac

olivierb2, 2020-05-24:
I'm affected as well, any hint to fix temporary this issue ? There is several files /lib/systemd/system/shadowsocks-libev* and I'm not sure which one I should alter to specify `LimitNOFILE=99999` and `LimitNPROC=99999`.

Dominaezzz, 2023-07-16:
I just installed openmptcprouter a couple days ago and I'm currently experiencing this issue.