Ysurac/openmptcprouter #814: Espressobin support

JorgeMoort89, 2020-01-31:
Hi Ysurac,
how can I make an image for Espressobin?
Do you have already tested it?

Ysurac, 2020-01-31:
You need to follow https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
No I've not tested it and it's a ARM A53 so not a really fast CPU.

JorgeMoort89, 2020-02-05:
On single connection I have very poor performance (40 mbit/s), how can I optimize it?
It have only 2 core 1.2Ghz

Ysurac, 2020-02-05:
You can try to change or disable encryption on shadowsocks.

JorgeMoort89, 2020-02-05:
How can I do that? I have to do it on server and OMR both? I need to reboot the router or the VPS?

Ysurac, 2020-02-05:
On the router in shadowsocks-libev interface.
But you will never get network high speed on slow CPU.

JorgeMoort89, 2020-02-05:
Without encryption is working great, have you seed the other issue about the public IP? Can you help me about it?

JorgeMoort89, 2020-02-18:
With ext4 filesystem when there are power loss I corrupt the OS, with squashfs I can't make any edit on configurations becouse the fs is in read only mode, how can I fix this?

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. Espressobin v7 is supported.
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac