Ysurac/openmptcprouter #815: VPN TUNNEL DOWN

Grynr38, 2020-01-31:
## Expected Behavior

VPN TUNNEL OK

## Actual Behavior

VPN TUNNEL DOWN

## Steps to Reproduce the Problem

  1. Install and configure VPS (OVH)
  2. Install and configure RPI 3B
  3. Wait for 24h

## Specifications

  - OpenMPTCProuter version: 0.53
  - OpenMPTCProuter VPS version: 0.1005
  - OpenMPTCProuter platform: RPI 3B

I have port-forwarding enabled and everything worked perfect during the first day... And today I have this problem. If I use OPENVPN instead, the tunnel works BUT portforwarding does not... and port forwarding is essential for me as I have small servers that I need to access from outside, cloudlessly.

I'll post the log in next post.
Thanks for everything

[image]
[image]

Grynr38, 2020-01-31:
Fri Jan 31 18:41:34 2020 daemon.err /usr/bin/ss-redir[14569]: remote recv: Operation timed out
Fri Jan 31 18:41:35 2020 daemon.err omr-tracker-ss[31471]: Failed to establish connection
Fri Jan 31 18:41:36 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Jan 31 18:41:38 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:41:39 2020 daemon.info omr-tracker-ss: Server (51.178.80.145) seems down, no answer to ping
Fri Jan 31 18:41:47 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:41:55 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:03 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:05 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:11 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:15 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Fri Jan 31 18:42:20 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:23 2020 daemon.err omr-tracker[31469]: Failed to establish connection
Fri Jan 31 18:42:28 2020 daemon.err /usr/bin/ss-local[14568]: getpeername: Socket not connected
Fri Jan 31 18:42:30 2020 user.notice omr-bypass: Restart dnsmasq...
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[12577]: exiting on receipt of SIGTERM
Fri Jan 31 18:42:30 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jan 31 18:42:30 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jan 31 18:42:30 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: started, version 2.80 cachesize 150
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: DNS service limited to local subnets
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain test
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain onion
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain localhost
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain local
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain invalid
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain bind
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using nameserver 127.0.0.1#5353
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: using local addresses only for domain lan
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: read /etc/hosts - 4 addresses
Fri Jan 31 18:42:30 2020 daemon.info dnsmasq[13813]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Jan 31 18:42:30 2020 user.notice omr-bypass: OMR-ByPass is running
Fri Jan 31 18:42:33 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 151.101.129.164)
Fri Jan 31 18:42:33 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Fri Jan 31 18:42:33 2020 user.notice Shadowsocks: Rules UP
Fri Jan 31 18:42:34 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Jan 31 18:42:34 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Jan 31 18:42:36 2020 user.notice post-tracking-post-tracking: wan1 switched off because ping from 192.168.50.200 error (80.67.169.12,80.67.169.40,4.2.2.1)
Fri Jan 31 18:42:36 2020 user.notice post-tracking-post-tracking: Delete vps (51.178.80.145) route via 192.168.50.202 dev wan1
Fri Jan 31 18:42:36 2020 user.notice post-tracking-post-tracking: wan1 down. Replace default route by 192.168.51.201 dev wan2
Fri Jan 31 18:42:37 2020 user.notice omr-bypass: Reload dnsmasq...
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[13813]: exiting on receipt of SIGTERM
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: started, version 2.80 cachesize 150
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: DNS service limited to local subnets
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain test
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain onion
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain localhost
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain local
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain invalid
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain bind
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using nameserver 127.0.0.1#5353
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: using local addresses only for domain lan
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: read /etc/hosts - 4 addresses
Fri Jan 31 18:42:37 2020 daemon.info dnsmasq[15144]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
[... cut ...]
Fri Jan 31 19:03:03 2020 user.notice omr-tracker: Launching...
Fri Jan 31 19:03:03 2020 user.notice omr-tracker: Launched
Fri Jan 31 19:03:11 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Fri Jan 31 19:03:19 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Fri Jan 31 19:03:31 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Fri Jan 31 19:03:31 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.50.202 dev wan1

Ysurac, 2020-01-31:
How do you change VPN ? Using the wizard ?

Grynr38, 2020-01-31:
I change them manualy : the wizard does not give me any option for changing them :

[image]

Grynr38, 2020-01-31:
The wizard does not give me any option for changing VPN; I change them manualy : 
quick question : is there a way to backup port forwarding settings ? (I have quite a few)
[image]

Grynr38, 2020-01-31:
When using OpenVPN it the tunnel seams fine, but port forwarding doesnt seem to work... And I quite prefer Glorytun... : 
[image]

Grynr38, 2020-01-31:
By the way, OPENVPN with **tunnel OK** and Glorytun with **tunnel down**; both give me the public VPN IP: 
[image]

Ysurac, 2020-01-31:
It's in advanced settings in the wizard and I think it's why you have problems changing the VPN.

Grynr38, 2020-01-31:
Oh ! Right, I did not noticed the advanced settings in the Wizard, thank you very much for the tip !!

I've just re-set Glorytun thru de wizard : 

[image]

But even thant way the tunnel is down : 
[image]

Ysurac, 2020-02-01:
Can you check the system log about glorytun  and make sure the another VPN is not running, and check log on the VPS too (/var/log/daemon.log on Debian), and make sure that glorytun is still running (```ps aux | grep glorytun```)

Grynr38, 2020-02-01:
Hi Ysurac and thanks for the follow up... Here are the results :
Glorytun grep : 

```
root     13548  0.0  0.0   2960   916 ?        Ss   Jan31   0:00 glorytun-tcp listener keyfile /etc/glorytun-tcp/tun0.key dev gt-tun0 port 65001 mptcp chacha20 retry count -1 const 5000000 timeout 10000 keepalive count 5 idle 20 interval 2 buffer-size 32768 multiqueue
root     24090  0.0  0.0   2692   900 ?        SLs  08:50   0:00 glorytun bind 0.0.0.0 65001 keyfile /etc/glorytun-udp/tun0.key dev gt-udp-tun0 chacha persist
root     24465  0.0  0.0   6148   820 pts/0    S+   08:53   0:00 grep glorytun
```
Seems fine. 

I can't find any other VPN running on the VPS log.

If there is way to backup the port forwarding, I would gladly restart everything from the begining...

Grynr38, 2020-02-01:
I did absolutely nothing, and everything came back as it should : 
[image]
I would really like to understand what is happening. If you let me, I'd like to leave this issue open for a few days in order to be able to report back if anything happens again.

Thanks for the wonderfull work, Ysurac.

Grynr38, 2020-02-03:
Sadly I abandonned this project, it is not reliable for a day to day use.
Thanks for the effort !

Ysurac, 2020-02-03:
Many users use it, and I use it. So it's reliable. It must be a configuration issue on your side

closed 2020-02-03 by Ysurac