Ysurac/openmptcprouter #816: Can't add additional WANs on Raspberry Pi 4

bnhf, 2020-02-01:
## Expected Behavior

Add WAN3 under Network - Interfaces with a physical adapter setting of eth1.5, set a static IP address, assign the interface to the WAN firewall group, and have the interface appear under OpenMPTCProuter - Status.

  In English please.

## Actual Behavior

After adding a 3rd WAN the interface becomes very slow, and more than one WAN3 is created under OpneMPTCProuter - Settings Wizard.  Attempting to add WAN3 via the Settings Wizard is also unsuccessful and must be assigned to a physical adapter as VLANs are not supported in the wizard.

## Steps to Reproduce the Problem

  1. Flash a fresh SD card
  2. Edit wan1 and wan2 under Network -Interfaces to set physical interfaces to eth1.4 and eth1.3
  3. Attempt to add a 3rd WAN under Network - Interfaces with eth1.5

## Specifications

  - OpenMPTCProuter version: v0.53.3 r0+11546-a74095c68c
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: RPi4

Ysurac, 2020-02-01:
LuCI become slow ? I would need system log of OpenMPTCProuter.

bnhf, 2020-02-01:
When I say slow, I mean like glacially slow -- I'm not sure I could even get logs through the GUI. Is there a way I could try to retrieve what you're looking for using WinSCP or SSH?

Ysurac, 2020-02-02:
Using logread via SSH

bnhf, 2020-02-02:
I've started with a fresh SD card to preserve my dual WAN setup that's working really well. After editing WAN1 and WAN2 to use eth1.4 and eth1.3 respectively, I've added a WAN3 that uses eth1.5 and noticed that the MAC address used is the same as WAN1 -- could this be the problem?

bnhf, 2020-02-02:
Now I'm wondering if macvlan isn't somehow getting in the way of using regular VLANs. I've been trying to setup a test environment to capture logs (when the RPi4 gets really slow), and I was able to change the physical interface for WAN1 to eth1.4, but when I tried to change WAN2 to eth1.3 the RPi4 became very, very slow even using SSH. I was able to capture the attached log however.

[OpenMPTCProuter_RPi4_log.txt](https://github.com/Ysurac/openmptcprouter/files/4144671/OpenMPTCProuter_RPi4_log.txt)

Ysurac, 2020-02-02:
I would need your config, System->OpenMPTCProuter, "all settings" tab or ```uci show``` via SSH.

closed 2020-05-18 by Ysurac