Ysurac/openmptcprouter #818: mptcp over vpn with obfuscation

BlueWaterCrystal, 2020-02-01:
MPTCP and UDP are blocked on the connection, unfortunately the connection requires http obfuscation for vpns

I've tired relaying openvpn udp via shadowsocks ss-tunnel but not working but does say packets are being filtered?

I've seen a guide people do openvpn over shadowsocks before but TCP over TCP

my guess is i haven't set it up right or its not possible ?

## Expected Behavior

  mptcp over openvpn udp relayed through ss-tunnel  shahowsocks

## Actual Behavior
wan1 and wan2:
shadowsocks with obfuscation works on the other connections that support mptcp
-----------------------------------

wan3:
shadowsocks with obfuscation works on the connection(no mptcp|udp) without mptcp
-----------------------------------
ovpnwan3:
no ip defined
no gateway defined
---------------------

## Steps to Reproduce the Problem

  1. #mptcp over vpn enabled

  2. #create ss_tunnel in shadowsocks:
-------------------------------
 server: SSS0
 local_address: 192.168.7.102
 local_port: 65301
 tunnel_address: 127.0.0.1:65301
 ipv6_first: 1
 fast_open: 1
 no_delay: 1
 reuse_port: 1
 mode: tcp_and_udp
 timeout: 1000
--------------------------------

3. OMR advanced settings:
--------------------------------
Enable ShadowSocks Obfuscating
Obfuscating plugin = simple-obfs
Obfuscating type = http
--------------------------------

4. shadowsocks remote servers:
--------------------------------
Plugin = obfs-local
Plugin Options =obfs=http;obfs-host=example.com
--------------------------------

## Specifications

  - OpenMPTCProuter version: (0.53)
  - OpenMPTCProuter VPS version: (0.1005)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-02-01:
I don't understand your configuration and why ss_tunnel can be used and you shouldn't have to change anything in shadowsocks config.
Also for obfuscation you should use v2ray.

BlueWaterCrystal, 2020-02-01:
mptcp over vpn keeps getting blocked and native connection doesn't support udp or mptcp, I'm not sure where to go from this.

Ysurac, 2020-02-01:
So MPTCP and UDP are blocked ? It's not a case i anticipated :)

closed 2020-02-03 by BlueWaterCrystal