Ysurac/openmptcprouter #819: Route public IP

JorgeMoort89, 2020-02-03:
Hi Ysurac,
on my vps I have assigned some public IP, how can I use them on the LAN of OMR for my clients?

For example:
Avaiable IPs:
188.1.1.2
188.1.1.14
188.1.1.17

I use 188.1.1.2 on my VPS to run OMR-VPS, I want to use 188.1.1.14. and 188.1.1.17 on my OMR LAN and assign that IP to my local clients. How can I do that?

Ysurac, 2020-02-05:
You may try to use them by configuring shorewall on the VPS but I never tested multiple public IP.

JorgeMoort89, 2020-02-05:
Mmm, in that way how can I use them in my OMR LAN?
Will be also useful a NAT 1:1 on that IPs.
Example: 192.168.100.200 -> 188.1.1.14

Ysurac, 2020-02-05:
You should be able to redirect them from the VPS to an IP on the LAN, but it you really want to attribute a public you will need a GRE tunnel.

JorgeMoort89, 2020-02-05:
How can I redirect them? Can you help me with a configuration example? I'll test it asap
Can I configure gre tunnel directly on OMR?

lirees, 2020-02-12:
I have the same problem, in my VPS i should manage a secondary public IP on my OMR lan

any solution?

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. VPS multiple IP is supported (I only need to add a doc now).
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac

Adorfer, 2020-08-15:
i am not sure if release 0.55 does cover the scenario in the initial request. 
At least as i understand the request to be different from e.g. #1006 
(in case i have a misunderstanding and the initial request is not what i describe in the following, i would create a new ticket, since it's a scenario i would like to cover)

senario 
--

- 188.1.1.1 is the primary v4 of the VPS.
- 188.1.1.2 is a public v4 on the VPS called "secondary"

At the local site of the OMR there is an applicance, which would like to use the 188.1.1.2 on it's interface. _"as if sitting in the rack of the vps provider"_
Why? Because it's an applicance where the setup does not allow a difference between "public IP" and "interface ip". Or where there is no adminstration login available ("managed appliance") 

What has to be done 
--

this may be done by putting up secondary LAN interface (physical or VLAN) on the OMR, 
something like 188.1.1.0/30.
it would probably involve some non-trivial fw-x-marking additional routing tables or the use of "network namespaces" (no idea if that's already available on the openwrt kernel)

Ysurac, 2020-08-15:
When multiples IP is available on the VPS, gre tunnel and shadowsocks ss-redir are created on OMR part. You then only have to configure shadowsocks rules to what you want like redirecting from an local IP/interface to the public IP you want. The output/input will be public IP, but the public IP can't be set directly to the appliance.

Adorfer, 2020-08-15:
And that would be my request: to use the (additional)  public IPv4s on the(another) LAN.

For the moment i put an additional router (hardware) as a nat device in between the OMR and the applicate to emulate that "provider-gateway". But it would be nice to get rid of those devices (here: UBNT-ERX)

Ysurac, 2020-08-15:
I don't understand what you want. You can create another LAN in OMR and set it to use the additional public IPv4 as input/ouput

Adorfer, 2020-08-15:
if "setting a fake public ip on a LAN-IF" is possible on the OMR: Perfekt. 
I was not aware, that this is possible now.