Ysurac/openmptcprouter #821: VPS open port

slitsevych, 2020-02-03:
## Expected Behavior
Be able to open port on VPS server without any redirection

## Actual Behavior
I cannot open one port (10050 for zabbix agent)  on VPS, connection is being timed out when I'm doing telnet to this port. 

## Steps to Reproduce the Problem

How can I open a port on VPS so that external service can communicate with it directly (maybe I need to adjust shorewall or something?)

Thank you in advance
## Specifications

  - OpenMPTCProuter version: 0.53.3
  - OpenMPTCProuter VPS version: 0.1008
  - OpenMPTCProuter platform: x86-64 (64-bit) ext4 image (EFI) installed on KVM machine

slitsevych, 2020-02-03:
it seems that I was able to bypass "redirection of all ports" and make an exception for two ports by modifying /etc/shorewall/rules and specifically by adjusting DNAT ports 

DNAT            net             vpn:$OMR_ADDR   tcp     1-10049
DNAT            net             vpn:$OMR_ADDR   tcp     10052-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-10049
DNAT            net             vpn:$OMR_ADDR   udp     10052-64999

closed 2020-02-04 by slitsevych

dettofatto, 2020-05-18:
So.. with this rule you have redirect only 2 ports to your vps? It is correct? You have done these settings on vps, or on router? Thank you!

slitsevych, 2020-05-18:
Hi! With these rules I redirect all ports except for two ports, 10050 and 10051. Settings are being set on vps, in shorewall config.
And i need this in order to be able to monitor router with zabbix internally without any vps involvement.

dettofatto, 2020-05-18:
Sorry for the stupid question... You have simply used nano /etc/shorewall/rules ?

dettofatto, 2020-05-18:
Ok... I've used a different way!
I've attached a screenshot

[image]

slitsevych, 2020-05-18:
Yeah, I think its the same and probably its more convenient to do this via GUI like you did.
In my case i was using nano, that's correct, and restart of shorewall service.

dettofatto, 2020-05-18:
In any case, with your settings, if the router go down, you can access to vps, in my case probably not...
I must investigate...