Ysurac/openmptcprouter #823: Pi-hole installation stalled on VPS with "Preparing to configure sudo (amd64)"

xabix, 2020-02-03:
## Expected Behavior

I want to install Pi-Hole on my VPS (Buster)

## Steps to Reproduce the Problem

I follow https://github.com/Ysurac/openmptcprouter/wiki/Pi-hole and select eht0, then Quad9 filtered, ipv4 only, 

[image]

My VPS runs on root so sudo is not by default activated. Could that be the issue? Maybe the VPS installation script should installed a local user to run OMR on the VPS on a local user versus root. But the VPS pihole installation script should know it as it is due to be installed just after.

Any idea of the issue? Should I first install sudo by myself?

## Specifications

  - OpenMPTCProuter version: 0.53.3
  - OpenMPTCProuter VPS version: Not visible anymore (cf screen shot but the latest)
[image]
  - OpenMPTCProuter platform: x86_64

xabix, 2020-02-03:
Welcome message on ssh on the VPS shows: 0.1007. Not sure if there is a newer version as it's not displayed anymore in the status screen as it used to be. Anyway not sure if this would solved the issue of Pihole installation.

xabix, 2020-02-03:
Isssue solved by doing:
`root@XabiX:~# dpkg --configure -a
Setting up php7.3-sqlite3 (7.3.11-1~deb10u1) ...

Creating config file /etc/php/7.3/mods-available/sqlite3.ini with new version

Creating config file /etc/php/7.3/mods-available/pdo_sqlite.ini with new version
Setting up php7.3-json (7.3.11-1~deb10u1) ...

Creating config file /etc/php/7.3/mods-available/json.ini with new version
Setting up php7.3-readline (7.3.11-1~deb10u1) ...

Creating config file /etc/php/7.3/mods-available/readline.ini with new version
Setting up php-sqlite3 (2:7.3+69) ...
Setting up sudo (1.8.27-1+deb10u1) ...

Configuration file '/etc/sudoers'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** sudoers (Y/I/N/O/D/Z) [default=N] ? D`

the sudo package should be forced to work seamlessly. I have also installed VPS 1.008 but still the status OMR webpage doesn't show anymore the version. Not sure why!

Thanks you may close this issue if no fix is needed on the pihole installation script.
merci

Ysurac, 2020-02-03:
sudo config was modified and it's why pihole hang. It ask the question if you want to keep it but it's not displayed by pihole install script.

1) I have no idea why you don't have the version displayed
2) I didn't add anything in resolv.conf (same as sudo I don't use it and didn't modify it via the script)
3)  You shouldn't need to reboot OpenMPTCProuter, I need to look at that, and I fixed the wiki thanks :)

xabix, 2020-02-03:
Thank you Ysurac!

closed 2020-02-03 by xabix