Ysurac/openmptcprouter #824: [WRT3200ACM] Max upload size prevents from uploading new image

jibsworld, 2020-02-04:
## Expected Behavior

It is possible to upload a new OMR image into the default WRT3200ACM admin UI

## Actual Behavior

When trying to upload a new system image >= 0.53 which is larger than 50MB, an error is displayed and then it is not possible to perform a new install

## Steps to Reproduce the Problem

  1. Go to the default router admin UI
  2. Go to "Connectivity" menu
  3. Try to upload a new OMR system image (>= 0.53)

## Specifications

  - OpenMPTCProuter version: 0.53.3
  - OpenMPTCProuter VPS version: not concerned
  - OpenMPTCProuter platform: Linksys WRT3200ACM

jibsworld, 2020-02-04:
There is a simple workaround:
- Install a 0.52.x (smaller than the integrated web server max up size)
- Upgrade it to the latest version

Possible resolution :
Reduce the image size to less than 50MB (limit seems to be defined exactly at 50MB after some tests and tries)

Ysurac, 2020-02-05:
It's a know issue, will be fixed in next release by removing some default installed packages.

closed 2020-05-18 by Ysurac