Ysurac/openmptcprouter #826: Bug ?

Slivikkus, 2020-02-04:
## Actual Behavior
J'ai l'impression d'avoir un bug
[image]

Pourtant j'ai bien la connexion au VPS, j'ignore si c'est normal ...
[image]

## Specifications

  - OpenMPTCProuter version: (0.53.3)
  - OpenMPTCProuter VPS version: (0.1008)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-02-04:
Il n'arrive pas à contacter le script d'administration et impossible d'obtenir une IP via shadowsocks, donc le traffic passe par le VPN qui est plus lent.
Il faut vérifier que le script d'admin est bien lancé sur le VPS, via ```ps aux | grep omr-admin``` sur le VPS et en vérifiant le /var/log/daemon.log si il s'agit d'une Debian.

Slivikkus, 2020-02-04:
[image]

Slivikkus, 2020-02-04:
J'ai fait l'essaie sur un autre VPS et ça fait exactement le même problème

Ysurac, 2020-02-05:
Il faut encore une fois vérifier sur le VPS ```/var/log/daemon.log``` si il s'agit d'une Debian.
Il faut vérifier également que la clef est correcte.
Si tout est bon, il me faudra la configuration du routeur présente dans Système->OpenMPTCProuter puis l'onglet voir tous les paramètres en supprimant les clefs.

Slivikkus, 2020-02-05:
Tout est ok, pour la clé j'ai copier coller ce que le VPS m'a donner, ça me fait pareil et peut importe quel VPS s'est, voici tout les paramètres de mon routeur

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='adky-vps-linux'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='00:0C:29:79:B0:CD'
dhcp.@host[0].ip='192.168.100.115'
dhcp.@host[1]=host
dhcp.@host[1].name='mail'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='E8:39:35:BC:55:84'
dhcp.@host[1].ip='192.168.100.129'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
[... 1343 lines cut ...]
unbound.@zone[1].zone_type='forward_zone'
unbound.@zone[1].zone_name='isp-bill.example.com.' 'isp-mail.example.net.'
upnpd.config=upnpd
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.enabled='1'
upnpd.config.uuid='b28d2d0a-150b-4178-ab92-788a043c2477'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth4' 'eth4' 'eth4' 'eth3'
Powered by openmptcprouter v0.53.3 r0+11546-a74095c68c

Slivikkus, 2020-02-05:
J'ai réinstaller OpenMPTCP mais ça n'a rien changer et côté VPS j'ai déjà réinstaller plusieurs fois mais ça fait la même chose :(

Ysurac, 2020-02-05:
Et le log ```/var/log/daemon.log``` ?

Slivikkus, 2020-02-05:
Y'a des tas d'information j'essaie de mettre ce qu'il y a ici ?

Ysurac, 2020-02-05:
ou sur pastebin.

Slivikkus, 2020-02-05:
ça va être compliquer j'arrive pas a accéder au SSH

Ysurac, 2020-02-05:
SSH est sur le port 65222, et si toujours pas d'accès ça expliquera le soucis.

Slivikkus, 2020-02-05:
[image]

Slivikkus, 2020-02-05:
ça me fait ça sur mes deux VPS quand je veux me connecté

Ysurac, 2020-02-05:
Si ils ont bien été redémarré ce n'est pas normal. Qui est le fournisseur ?

Slivikkus, 2020-02-05:
OVH et Sapinet

Ysurac, 2020-02-05:
En testant sur un VPS OVH sous Debian 10 Buster je n'ai pas eu de soucis pour l'accès SSH après reboot ni pour l'utiliser après configuration d'OpenMPTCProuter.

Slivikkus, 2020-02-05:
Je pense que c'est à cause de la configuration de l'IPv6 je continu de regader

closed 2020-02-05 by Slivikkus