Ysurac/openmptcprouter #832: No answer (impossible de contacter le script d'administration du serveur)

CoSysopK, 2020-02-08:
## Actual Behavior
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:16:57 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Host is unreachable
Sat Feb  8 18:16:59 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:16:59 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:16:59 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:16:59 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:17:00 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:00 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:01 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:02 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:02 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:17:02 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:03 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:05 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Host is unreachable
Sat Feb  8 18:17:05 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:17:06 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Host is unreachable
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:17:07 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:08 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:08 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:17:09 2020 daemon.err /usr/bin/ss-redir[11404]: remote recv: Operation timed out
Sat Feb  8 18:17:10 2020 daemon.err /usr/bin/ss-local[11403]: getpeername: Socket not connected
Sat Feb  8 18:17:11 2020 daemon.err /usr/bin/ss-redir[11407]: remote recv: Operation timed out
Sat Feb  8 18:17:18 2020 daemon.err /usr/bin/ss-local[11403]: getpeername: Socket not connected
Sat Feb  8 18:17:19 2020 daemon.err /usr/bin/ss-redir[11405]: remote recv: Operation timed out
Sat Feb  8 18:17:19 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:17:19 2020 daemon.err /usr/bin/ss-redir[11406]: remote recv: Operation timed out
Sat Feb  8 18:17:26 2020 daemon.err /usr/bin/ss-local[11403]: getpeername: Socket not connected
Sat Feb  8 18:17:26 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 88.191.250.176, 151.101.129.164, 1.1.1.1)
Sat Feb  8 18:17:26 2020 user.notice Shadowsocks: Rules DOWN
Sat Feb  8 18:17:32 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sat Feb  8 18:17:35 2020 user.notice OMR-VPS: Can't get token, try later
Sat Feb  8 18:17:35 2020 daemon.err /usr/bin/ss-local[11403]: getpeername: Socket not connected
[image]

[image]

Thank you for your information.

## Steps to Reproduce the Problem

1. Installer l'image pour Raspberry PI 3 / 3B + (64 bits)
2. Installer VPS Debian 10 Buster x86_64
3. Configurez le routeur avec le wan1 statique et le wan2 statique
4. Ajoutez l'adresse IP VPS et, toutes les informations récupérées sur le VPS
5. Deux clés 4G sur USB PI 3B

## Specifications

- Version OpenMPTCProuter: version 0.53.3
- Version VPS OpenMPTCProuter: Version 0.1008
- Plateforme OpenMPTCProuter: RPI3B + 64 bits

Ysurac, 2020-02-08:
Inutile de configurer les clefs dans les paramètres avancés et le VPN doit pas être sur none.
Et il y a certainement d'autres erreurs de configuration.

closed 2020-02-08 by CoSysopK

Ysurac, 2020-02-08:
c'est corrigé ?

CoSysopK, 2020-02-08:
I'm going to open a new post because I did a new installation of the router