Ysurac/openmptcprouter #833: VPN no answer

CoSysopK, 2020-02-08:
## Actual Behavior

I just did a new reinstall of the router. After entering the necessary information, here are the results I get !!!
[image]

For information, I have not restarted the router, since the new installation, about 1 hour ago.

## Steps to Reproduce the Problem

1. Installer l'image pour Raspberry PI 3 / 3B + (64 bits)
2. Installateur VPS Debian 10 Buster x86_64
3. Configurez le routeur avec le wan1 statique et le wan2 statique
4. Ajoutez l'adresse IP VPS et, toutes les informations récupérées sur le VPS
5. Deux clés 4G sur USB PI 3B

## Specifications

- Version OpenMPTCProuter: version 0.53.3
- Version VPS OpenMPTCProuter: Version 0.1008
- Plateforme OpenMPTCProuter: RPI3B + 64 bits

CoSysopK, 2020-02-08:
I just had these errors:

Sat Feb  8 19:24:19 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Feb  8 19:24:19 2020 daemon.info dnsmasq[23920]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sat Feb  8 19:24:19 2020 daemon.info dnsmasq-dhcp[23920]: read /etc/ethers - 0 addresses
Sat Feb  8 19:25:34 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:25:35 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:25:35 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:25:36 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:25:36 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:25:37 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:25:38 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:25:38 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:25:38 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:25:39 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:25:39 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:25:39 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:25:40 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:25:41 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:25:41 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:28:09 2020 daemon.err /usr/bin/ss-redir[32238]: server recv: Operation timed out
Sat Feb  8 19:32:21 2020 daemon.info hostapd: wlan0: STA 70:bb:e9:c5:0f:f7 WPA: group key handshake completed (RSN)
Sat Feb  8 19:34:10 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:34:10 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:34:11 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:34:12 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:34:12 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:34:15 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:34:16 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:34:17 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:34:17 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:34:18 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:34:18 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:34:18 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:34:19 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:34:19 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:34:19 2020 daemon.info glorytun[32339]: STOPPED tun0

CoSysopK, 2020-02-08:
Sat Feb  8 19:34:19 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:42:21 2020 daemon.info hostapd: wlan0: STA 70:bb:e9:c5:0f:f7 WPA: group key handshake completed (RSN)
Sat Feb  8 19:42:45 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:42:46 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:42:46 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:42:47 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:42:48 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:42:49 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:42:50 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:42:50 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:42:50 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:42:51 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:42:51 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:42:51 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:42:52 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:42:53 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:42:53 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:22 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:22 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:22 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:23 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:23 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:31 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:32 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:33 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:33 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:34 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:34 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:34 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:35 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:35 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:35 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:36 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:37 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:37 2020 daemon.info glorytun[32339]: STOPPED tun0
Sat Feb  8 19:51:38 2020 daemon.err glorytun[32339]: 34.77.160.206.65001: connected
Sat Feb  8 19:51:38 2020 daemon.info glorytun[32339]: STARTED tun0
Sat Feb  8 19:51:38 2020 daemon.info glorytun[32339]: STOPPED tun0

Ysurac, 2020-02-08:
Les clefs sont normalement récupéré via l'API sur le VPS après validation du wizard.
Tu peux essayer de revalider le wizard.

CoSysopK, 2020-02-08:
Always "le VPN ne répond pas"

[image]

CoSysopK, 2020-02-08:
Sat Feb  8 20:07:26 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Connection reset by peer
Sat Feb  8 20:08:33 2020 daemon.err glorytun[1980]: 34.77.160.206.65001: connected
Sat Feb  8 20:08:33 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 20:08:33 2020 daemon.info glorytun[1980]: STOPPED tun0
Sat Feb  8 20:08:34 2020 daemon.err glorytun[1980]: 34.77.160.206.65001: connected
Sat Feb  8 20:08:35 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 20:08:35 2020 daemon.info glorytun[1980]: STOPPED tun0
Sat Feb  8 20:08:36 2020 daemon.err glorytun[1980]: 34.77.160.206.65001: connected
Sat Feb  8 20:08:36 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 20:08:36 2020 daemon.info glorytun[1980]: STOPPED tun0
Sat Feb  8 20:08:37 2020 daemon.err glorytun[1980]: 34.77.160.206.65001: connected
Sat Feb  8 20:08:37 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 20:08:38 2020 daemon.info glorytun[1980]: STOPPED tun0
Sat Feb  8 20:08:39 2020 daemon.err glorytun[1980]: 34.77.160.206.65001: connected
Sat Feb  8 20:08:39 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 20:08:39 2020 daemon.info glorytun[1980]: STOPPED tun0
Sat Feb  8 20:09:53 2020 user.notice post-tracking-post-tracking: Change master interface from wan2 (67 ms) to wan1 (40 ms)
Sat Feb  8 20:10:01 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.101.1 dev eth1

Ysurac, 2020-02-08:
Pas la peine de me copier plusieurs fois la même erreur...
Vérifie si la clef est bien présente pour glorytun et sur le VPS dans /var/log/daemon.log ce qu'il peut y avoir à propos de glorytun.

CoSysopK, 2020-02-08:
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Host is unreachable
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:24 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:25 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:25 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:25 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:25 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:25 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Host is unreachable
Sat Feb  8 20:12:27 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:27 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Host is unreachable
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:28 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Host is unreachable
Sat Feb  8 20:12:29 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Host is unreachable
Sat Feb  8 20:12:30 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:31 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:31 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:31 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-local[1910]: getpeername: Socket not connected
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:32 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:34 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:34 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:34 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Host is unreachable
Sat Feb  8 20:12:35 2020 user.notice post-tracking-post-tracking: Change master interface from wan1 (152 ms) to wan2 (36 ms)
Sat Feb  8 20:12:36 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:36 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:36 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:37 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Host is unreachable
Sat Feb  8 20:12:42 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Host is unreachable
Sat Feb  8 20:12:42 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Host is unreachable
Sat Feb  8 20:12:42 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Host is unreachable
Sat Feb  8 20:12:42 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Host is unreachable
Sat Feb  8 20:12:43 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Host is unreachable
Sat Feb  8 20:12:43 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Host is unreachable
Sat Feb  8 20:12:43 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Host is unreachable
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Host is unreachable
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1913]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:44 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
[... cut ...]
Sat Feb  8 20:12:50 2020 daemon.err /usr/bin/ss-redir[1912]: remote recv: Operation timed out
Sat Feb  8 20:12:52 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:52 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Host is unreachable
Sat Feb  8 20:12:55 2020 daemon.err /usr/bin/ss-local[1910]: getpeername: Socket not connected
Sat Feb  8 20:12:56 2020 daemon.err /usr/bin/ss-redir[1914]: remote recv: Operation timed out
Sat Feb  8 20:12:56 2020 daemon.err /usr/bin/ss-redir[1911]: remote recv: Operation timed out

CoSysopK, 2020-02-08:
Glorytun key is present on the router.

CoSysopK, 2020-02-08:
Feb  8 19:25:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 19:25:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 19:25:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 1.
Feb  8 19:25:45 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 19:25:45 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 19:25:45 instance-1 systemd-udevd[12894]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 19:25:45 instance-1 systemd-networkd[12898]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 19:25:46 instance-1 glorytun-tcp-run[12938]: 185.228.228.103.54067: connected
Feb  8 19:25:46 instance-1 glorytun-tcp-run[12938]: tun write: Input/output error
Feb  8 19:25:46 instance-1 glorytun-tcp-run[12938]: INITIALIZED gt-tun0
Feb  8 19:25:46 instance-1 glorytun-tcp-run[12938]: STARTED gt-tun0
Feb  8 19:25:46 instance-1 glorytun-tcp-run[12938]: STOPPED gt-tun0
Feb  8 19:25:46 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 19:25:46 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 19:25:46 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 2.
Feb  8 19:25:46 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 19:25:46 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 19:25:46 instance-1 systemd-udevd[12894]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 19:25:46 instance-1 systemd-networkd[12898]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 19:25:47 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 19:25:47 instance-1 glorytun-tcp-run[12946]: 185.228.228.103.33770: connected
Feb  8 19:25:47 instance-1 glorytun-tcp-run[12946]: tun write: Input/output error
Feb  8 19:25:47 instance-1 glorytun-tcp-run[12946]: INITIALIZED gt-tun0
Feb  8 19:25:47 instance-1 glorytun-tcp-run[12946]: STARTED gt-tun0
Feb  8 19:25:47 instance-1 glorytun-tcp-run[12946]: STOPPED gt-tun0
Feb  8 19:25:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 19:25:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 19:25:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 3.
Feb  8 19:25:47 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 19:25:47 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 19:25:47 instance-1 systemd-udevd[12894]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 19:25:48 instance-1 systemd-networkd[12898]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 19:25:48 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 19:25:48 instance-1 glorytun-tcp-run[12956]: 185.228.228.103.7540: connected
Feb  8 19:25:49 instance-1 glorytun-tcp-run[12956]: tun write: Input/output error
Feb  8 19:25:49 instance-1 glorytun-tcp-run[12956]: INITIALIZED gt-tun0
Feb  8 19:25:49 instance-1 glorytun-tcp-run[12956]: STARTED gt-tun0
Feb  8 19:25:49 instance-1 glorytun-tcp-run[12956]: STOPPED gt-tun0
Feb  8 19:25:49 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 19:25:49 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 19:25:49 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 4.
Feb  8 19:25:49 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 19:25:49 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 19:25:49 instance-1 systemd-udevd[12894]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 19:25:49 instance-1 systemd-networkd[12898]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 19:25:50 instance-1 glorytun-tcp-run[12964]: 185.228.228.103.57057: connected
Feb  8 19:25:50 instance-1 glorytun-tcp-run[12964]: tun write: Input/output error
Feb  8 19:25:50 instance-1 glorytun-tcp-run[12964]: INITIALIZED gt-tun0
Feb  8 19:25:50 instance-1 glorytun-tcp-run[12964]: STARTED gt-tun0
Feb  8 19:25:50 instance-1 glorytun-tcp-run[12964]: STOPPED gt-tun0
Feb  8 19:25:50 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 19:25:50 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 19:25:50 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 5.
Feb  8 19:25:50 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 19:25:50 instance-1 systemd[1]: glorytun-tcp@tun0.service: Start request repeated too quickly.
Feb  8 19:25:50 instance-1 systemd[1]: glorytun-tcp@tun0.service: Failed with result 'start-limit-hit'.
Feb  8 19:25:50 instance-1 systemd[1]: Failed to start Glorytun TCP on tun0.
Feb  8 19:25:50 instance-1 systemd[1]: Created slice User Slice of UID 1000.
Feb  8 19:25:50 instance-1 systemd[1]: Starting User Runtime Directory /run/user/1000...
Feb  8 19:25:50 instance-1 systemd[1]: Started User Runtime Directory /run/user/1000.

Ysurac, 2020-02-08:
The problem is : " tun write: Input/output error"
What is VPS provider ? It's a Debian ?

CoSysopK, 2020-02-08:
yes.
Cloud Google with Debian 10

CoSysopK, 2020-02-08:
I get the impression that this error repeats about every 8 minutes.

Ysurac, 2020-02-08:
Never tested on Cloud Google.
What is the result of "uname -a" on the VPS ?

CoSysopK, 2020-02-08:
Linux instance-1 4.19.80-mptcp #1 SMP Wed Nov 20 12:28:05 UTC 2019 x86_64 GNU/Linux

Ysurac, 2020-02-08:
ok
I think the problem is that gt-tun0 interface doesn't get an IP. This should be done via systemd.
Can you check that the file ```/lib/systemd/network/glorytun-tcp.network``` exist ?
if yes can do a ```systemctl restart systemd-networkd``` and a ```systemctl restart glorytun-tcp@tun0``` and check log again ?

CoSysopK, 2020-02-08:
/lib/systemd/network/glorytun-tcp.network   existe

Feb  8 20:12:43 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 20:12:43 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 20:12:43 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 4.
Feb  8 20:12:43 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:43 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 20:12:43 instance-1 systemd-udevd[19522]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 20:12:43 instance-1 systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 20:12:44 instance-1 glorytun-tcp-run[19551]: 185.228.228.103.54667: connected
Feb  8 20:12:44 instance-1 glorytun-tcp-run[19551]: tun write: Input/output error
Feb  8 20:12:44 instance-1 glorytun-tcp-run[19551]: INITIALIZED gt-tun0
Feb  8 20:12:44 instance-1 glorytun-tcp-run[19551]: STARTED gt-tun0
Feb  8 20:12:44 instance-1 glorytun-tcp-run[19551]: STOPPED gt-tun0
Feb  8 20:12:44 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 20:12:44 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 20:12:44 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 5.
Feb  8 20:12:44 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:44 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 20:12:44 instance-1 systemd-udevd[19522]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 20:12:44 instance-1 systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 20:12:45 instance-1 glorytun-tcp-run[19561]: 185.228.228.103.12641: connected
Feb  8 20:12:45 instance-1 glorytun-tcp-run[19561]: tun write: Input/output error
Feb  8 20:12:45 instance-1 glorytun-tcp-run[19561]: INITIALIZED gt-tun0
Feb  8 20:12:45 instance-1 glorytun-tcp-run[19561]: STARTED gt-tun0
Feb  8 20:12:45 instance-1 glorytun-tcp-run[19561]: STOPPED gt-tun0
Feb  8 20:12:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 20:12:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 20:12:45 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 6.
Feb  8 20:12:45 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:45 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 20:12:46 instance-1 systemd-udevd[19522]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 20:12:46 instance-1 systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 20:12:46 instance-1 glorytun-tcp-run[19571]: 185.228.228.103.12640: connected
Feb  8 20:12:47 instance-1 glorytun-tcp-run[19571]: tun write: Input/output error
Feb  8 20:12:47 instance-1 glorytun-tcp-run[19571]: INITIALIZED gt-tun0
Feb  8 20:12:47 instance-1 glorytun-tcp-run[19571]: STARTED gt-tun0
Feb  8 20:12:47 instance-1 glorytun-tcp-run[19571]: STOPPED gt-tun0
Feb  8 20:12:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 20:12:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 20:12:47 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 7.
Feb  8 20:12:47 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:47 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 20:12:47 instance-1 systemd-udevd[19522]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 20:12:47 instance-1 systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 20:12:48 instance-1 glorytun-tcp-run[19581]: 185.228.228.103.46390: connected
Feb  8 20:12:48 instance-1 glorytun-tcp-run[19581]: tun write: Input/output error
Feb  8 20:12:48 instance-1 glorytun-tcp-run[19581]: INITIALIZED gt-tun0
Feb  8 20:12:48 instance-1 glorytun-tcp-run[19581]: STARTED gt-tun0
Feb  8 20:12:48 instance-1 glorytun-tcp-run[19581]: STOPPED gt-tun0
Feb  8 20:12:48 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 20:12:48 instance-1 systemd[1]: glorytun-tcp@tun0.service: Service RestartSec=100ms expired, scheduling restart.
Feb  8 20:12:48 instance-1 systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 8.
Feb  8 20:12:48 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:48 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 20:12:48 instance-1 systemd-udevd[19522]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 20:12:48 instance-1 systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 20:12:49 instance-1 glorytun-tcp-run[19596]: 185.228.228.103.54069: connected
Feb  8 20:12:49 instance-1 glorytun-tcp-run[19596]: tun write: Input/output error
Feb  8 20:12:49 instance-1 glorytun-tcp-run[19596]: INITIALIZED gt-tun0
Feb  8 20:12:49 instance-1 glorytun-tcp-run[19596]: STARTED gt-tun0
Feb  8 20:12:49 instance-1 glorytun-tcp-run[19596]: STOPPED gt-tun0

[... cut ...]
Feb  8 20:12:49 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 20:12:49 instance-1 systemd[1]: glorytun-tcp@tun0.service: Start request repeated too quickly.
Feb  8 20:12:49 instance-1 systemd[1]: glorytun-tcp@tun0.service: Failed with result 'start-limit-hit'.
Feb  8 20:12:49 instance-1 systemd[1]: Failed to start Glorytun TCP on tun0.
Feb  8 20:13:15 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 20:13:46 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected

CoSysopK, 2020-02-08:
Sat Feb  8 21:17:26 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 21:17:26 2020 daemon.info glorytun[1980]: STOPPED tun0

I get the impression that this error repeats about every 8 minutes.

Ysurac, 2020-02-08:
It's because it can write to tun interface.
This may be related to " systemd-networkd[19457]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory"

CoSysopK, 2020-02-08:
i can disable ipv6, i don't mind

Ysurac, 2020-02-08:
You can try. 
I will also try to get a Cloud Google maybe tomorrow.

CoSysopK, 2020-02-08:
[image]

CoSysopK, 2020-02-08:
I disabled ipv6 and always :
Sat Feb  8 21:43:11 2020 daemon.info glorytun[1980]: STARTED tun0
Sat Feb  8 21:43:11 2020 daemon.info glorytun[1980]: STOPPED tun0

Ysurac, 2020-02-08:
The 0B is logical as it's not working. And the issue is not the "STARTED/STOPPED" but the tun write error.

Ysurac, 2020-02-08:
after you disabled IPv6, did you do ```systemctl restart systemd-networkd``` and a ```systemctl restart glorytun-tcp@tun0``` ?

CoSysopK, 2020-02-08:
no, i do

CoSysopK, 2020-02-08:
always the same error

Ysurac, 2020-02-08:
ok, I will try to get a cloud Google tomorrow or beginning of next week.

CoSysopK, 2020-02-08:
I do tests with glorytun-udp and dsvpn

CoSysopK, 2020-02-08:
With glorytun-udp:

[image]
Sat Feb  8 22:00:05 2020 daemon.err /usr/bin/ss-redir[17685]: remote recv: Operation timed out
Sat Feb  8 22:00:07 2020 daemon.err /usr/bin/ss-redir[17686]: remote recv: Operation timed out
Sat Feb  8 22:00:07 2020 daemon.err /usr/bin/ss-redir[17687]: remote recv: Operation timed out
Sat Feb  8 22:00:08 2020 daemon.err /usr/bin/ss-redir[17685]: remote recv: Operation timed out
Sat Feb  8 22:00:08 2020 daemon.err /usr/bin/ss-redir[17684]: remote recv: Operation timed out
Sat Feb  8 22:00:08 2020 daemon.err /usr/bin/ss-redir[17685]: remote recv: Operation timed out
Sat Feb  8 22:00:08 2020 user.notice OMR-VPS: Can't get token, try later
Sat Feb  8 22:00:09 2020 daemon.err /usr/bin/ss-redir[17687]: remote recv: Host is unreachable
Sat Feb  8 22:00:09 2020 daemon.err /usr/bin/ss-local[17683]: getpeername: Socket not connected
Sat Feb  8 22:00:09 2020 daemon.err /usr/bin/ss-redir[17686]: remote recv: Operation timed out
Sat Feb  8 22:00:09 2020 daemon.err /usr/bin/ss-redir[17684]: remote recv: Operation timed out
Sat Feb  8 22:00:11 2020 daemon.err /usr/bin/ss-redir[17687]: remote recv: Operation timed out
Sat Feb  8 22:00:13 2020 daemon.err /usr/bin/ss-redir[17686]: remote recv: Operation timed out
Sat Feb  8 22:00:13 2020 user.notice OMR-VPS: Can't get token, try later
Sat Feb  8 22:00:14 2020 daemon.err /usr/bin/ss-redir[17684]: remote recv: Operation timed out
Sat Feb  8 22:00:14 2020 daemon.err /usr/bin/ss-redir[17686]: remote recv: Operation timed out
Sat Feb  8 22:00:24 2020 daemon.err /usr/bin/ss-redir[17686]: remote recv: Host is unreachable
Sat Feb  8 22:00:26 2020 daemon.err /usr/bin/ss-redir[17684]: remote recv: Operation timed out

and 18 minutes without log on the router.

On the VPS (/var/log/daemon.log):
Feb  8 21:09:05 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:09:37 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:10:08 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:10:40 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:10:40 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 21:11:01 instance-1 dhclient[298]: DHCPREQUEST for 10.132.0.2 on ens4 to 169.254.169.254 port 67
Feb  8 21:11:01 instance-1 dhclient[298]: DHCPACK of 10.132.0.2 from 169.254.169.254
Feb  8 21:11:01 instance-1 dhclient[298]: bound to 10.132.0.2 -- renewal in 1372 seconds.
Feb  8 21:11:12 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:11:44 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:12:15 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:12:15 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:13:57 instance-1 ss-server[387]: crypto: AEAD: repeat salt detected
Feb  8 21:17:28 instance-1 systemd[1]: Stopping Glorytun UDP on tun0...
Feb  8 21:17:28 instance-1 systemd[1]: glorytun-udp@tun0.service: Succeeded.
Feb  8 21:17:28 instance-1 systemd[1]: Stopped Glorytun UDP on tun0.
Feb  8 21:17:28 instance-1 systemd[1]: Started Glorytun UDP on tun0.
Feb  8 21:17:28 instance-1 glorytun-udp-run[1924]: running on device gt-udp-tun0 as pid 1924
Feb  8 21:17:28 instance-1 systemd[1]: Stopping Network Service...
Feb  8 21:17:28 instance-1 systemd[1]: systemd-networkd.service: Succeeded.
Feb  8 21:17:28 instance-1 systemd[1]: Stopped Network Service.
Feb  8 21:17:28 instance-1 systemd[1]: Starting Network Service...
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: dsvpn0: Gained IPv6LL
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: omr-6in4: Gained IPv6LL
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: tun0: Gained IPv6LL
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: tun1: Gained IPv6LL
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: ens4: Gained IPv6LL
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: Enumeration completed
Feb  8 21:17:28 instance-1 systemd[1]: Started Network Service.
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: gt-udp-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:17:28 instance-1 systemd-networkd[1928]: mlvpn0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:18:08 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 21:18:10 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 21:18:13 instance-1 ss-server[387]: remote recv: Connection reset by peer
Feb  8 21:18:14 instance-1 ss-server[387]: remote recv: Connection reset by peer

i keep testing with glorytun-udp to see the logs.

Ysurac, 2020-02-08:
Same problem.

CoSysopK, 2020-02-08:
OK so I test with dsvpn.

CoSysopK, 2020-02-08:
After choosing dsvpn and save and apply in the wizard:
[image]
[image]
No log on the router.

CoSysopK, 2020-02-08:
here the log on the VPS:
Feb  8 21:31:10 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:10 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:12 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:13 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:20 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:29 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:42 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:42 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:43 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:43 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:43 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:31:44 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:00 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:15 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:20 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:32 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:52 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:32:58 instance-1 systemd[1]: Stopping Network Service...
Feb  8 21:32:58 instance-1 systemd[1]: systemd-networkd.service: Succeeded.
Feb  8 21:32:58 instance-1 systemd[1]: Stopped Network Service.
Feb  8 21:32:58 instance-1 systemd[1]: Starting Network Service...
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: dsvpn0: Gained IPv6LL
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: omr-6in4: Gained IPv6LL
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: tun0: Gained IPv6LL
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: tun1: Gained IPv6LL
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: ens4: Gained IPv6LL
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: Enumeration completed
Feb  8 21:32:58 instance-1 systemd[1]: Started Network Service.
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: gt-udp-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:32:58 instance-1 systemd-networkd[6434]: mlvpn0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:32:58 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:33:01 instance-1 glorytun-tcp-run[31131]: INITIALIZED gt-tun0
Feb  8 21:33:01 instance-1 systemd[1]: Stopping Glorytun TCP on tun0...
Feb  8 21:33:01 instance-1 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb  8 21:33:01 instance-1 systemd[1]: Stopped Glorytun TCP on tun0.
Feb  8 21:33:01 instance-1 systemd[1]: Started Glorytun TCP on tun0.
Feb  8 21:33:01 instance-1 systemd-udevd[6439]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb  8 21:33:01 instance-1 systemd-networkd[6434]: gt-tun0: DHCP6 CLIENT: Failed to set DUID: No such file or directory
Feb  8 21:33:09 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:33:41 instance-1 ss-server[387]: getpeername: Transport endpoint is not connected
Feb  8 21:33:53 instance-1 dhclient[298]: DHCPREQUEST for 10.132.0.2 on ens4 to 169.254.169.254 port 67
Feb  8 21:33:53 instance-1 dhclient[298]: DHCPACK of 10.132.0.2 from 169.254.169.254
Feb  8 21:33:53 instance-1 dhclient[298]: bound to 10.132.0.2 -- renewal in 1503 seconds.

Ysurac, 2020-02-08:
possible that the issue with glorytun make everything fail.

CoSysopK, 2020-02-09:
with dsvpn, today there was a problem:

Sun Feb  9 19:06:25 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.101.1 dev eth1
Sun Feb  9 19:06:25 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun Feb  9 19:06:25 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Sun Feb  9 19:06:26 2020 user.notice Shadowsocks: Rules DOWN
Sun Feb  9 19:06:27 2020 user.notice Shadowsocks: Rules UP
Sun Feb  9 19:06:27 2020 daemon.err omr-tracker[760]: RTNETLINK answers: Permission denied
Sun Feb  9 19:06:28 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Feb  9 19:06:28 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb  9 19:06:28 2020 daemon.err omr-tracker[760]: 0+5 records in
Sun Feb  9 19:06:28 2020 daemon.err omr-tracker[760]: 0+5 records out
Sun Feb  9 19:06:28 2020 daemon.err omr-tracker[760]: 0+5 records in
Sun Feb  9 19:06:28 2020 daemon.err omr-tracker[760]: 0+5 records out
Sun Feb  9 19:06:30 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Feb  9 19:06:31 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.102.1 dev eth2
Sun Feb  9 19:06:31 2020 daemon.info dnsmasq[7335]: read /etc/hosts - 4 addresses
Sun Feb  9 19:06:31 2020 daemon.info dnsmasq[7335]: read /tmp/hosts/odhcpd - 0 addresses
Sun Feb  9 19:06:31 2020 daemon.info dnsmasq[7335]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Feb  9 19:06:31 2020 daemon.info dnsmasq-dhcp[7335]: read /etc/ethers - 0 addresses
Sun Feb  9 19:06:31 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Feb  9 19:06:38 2020 daemon.err uhttpd[7995]: luci: accepted login on / for root from 192.168.100.228

[image]

CoSysopK, 2020-02-10:
Today I discovered this log.

Mon Feb 10 10:26:22 2020 daemon.err /usr/bin/ss-local[28201]: getpeername: Socket not connected
Mon Feb 10 10:26:42 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.101.1 dev eth1
Mon Feb 10 10:26:42 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Mon Feb 10 10:26:42 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Mon Feb 10 10:26:42 2020 user.notice Shadowsocks: Rules DOWN
Mon Feb 10 10:26:43 2020 user.notice Shadowsocks: Rules UP
**Mon Feb 10 10:26:43 2020 daemon.err omr-tracker-ss[761]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?**
Mon Feb 10 10:26:43 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules

Ysurac, 2020-02-10:
I can reproduce the issue on Google Cloud. IP are not distributed to VPN. I will try to fix this strange issue asap.

Ysurac, 2020-02-10:
Seems to be fixed after an ```apt-get update``` and ```apt-get upgrade``` that update systemd.

CoSysopK, 2020-02-10:
For me, it's:

The following packages have been kept back:
  shadowsocks-libev
The following packages will be upgraded:
  google-cloud-sdk google-osconfig-agent
2 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

I did the update.

CoSysopK, 2020-02-10:
glorytun tcp very nice 👍

CoSysopK, 2020-02-10:
[image]

is it important to make these updates?

CoSysopK, 2020-02-10:
I ask this question, to know if I can make these updates if however there is a problem.

Ysurac, 2020-02-10:
You shouldn't update OpenMPTCProuter this way. It's not supported or tested for now.

CoSysopK, 2020-02-10:
I continue the test with glorytun tcp. and then i will test glorytun udp then if it works carefree i will close the post.

CoSysopK, 2020-02-11:
Today, I experienced instability with Glorytun tcp. Which you will see in the picture.

[image]

I will test Glorytun udp, since the udp protocol is faster compared to the tcp protocol.

Would it be possible to deactivate the usb port where there was a red problem with the WAN2 and reactivate it so that the 4g usb stick restarts without having to unplug it and then plug it back in.
Because, I had to unplug it and plug it back in to make it operational.
When there is this type of error with a WAN, the router should take action to repair this error.
Often this error is corrected after a while, but it takes several minutes. So, if there is a problem with WAN1, it will no longer be able to have link aggregation since WAN2 is HS.

closed 2020-02-14 by CoSysopK