Ysurac/openmptcprouter #834: Problème IP (VPS OVH)

Slivikkus, 2020-02-09:
## Actual Behavior
Bonjour, encore le problème, j'ai repris un VPS chez OVH config IPv6 fonctionnelle mais ça fonctionnais pas avant que j'ajoute la config IPv6

J'ai fait la commande `s aux | grep omr-admin` j'ai obtenu le même résultat 

j'ai décidément pas de chance :-( 

## Specifications

  - OpenMPTCProuter version: (0.53.3)
  - OpenMPTCProuter VPS version: (0.1008)
  - OpenMPTCProuter platform: (x86_64)

dabuide, 2020-02-11:
Petite question bête mais une fois que tu as installé le script sur le serveur OVH est-ce que tu utilises le KVM pour redémarrer la VM et ainsi choisir la partition MPTCP sur Debian ?

Slivikkus, 2020-02-11:
Oui et pourtant je suis bien sur l'adresse ip du VPS, mais ça m'affiche ça, mais dans tout les cas la partition MPTPC est mise par défaut sur MPTCP donc ça viendrais pas de là, de plus j'ai des lenteurs au niveau de la connexion par rapport à d'habitude

dabuide, 2020-02-11:
Non c'est bien là le problème il ne se lance pas par défaut ! En tout cas quand j'ai pris mon serveur OVH j'ai du le faire à la main il suffit d'accéder en SSH sur le VPS et de faire un "uname -a" afin de vérifier que le kernel est bien MPTCP

Slivikkus, 2020-02-11:
Bah moi j'ai bien regarder le moment de démarrage et il le sélectionne bien

dabuide, 2020-02-11:
Après pour l'adresse IPv6 je ne me suis pas attardé dessus cela ne fonctionne pas correctement du coup j'ai laissé en IPv4 et aucun souci. En tout cas est-ce que l'accès en SSH via le port 65222 fonctionne ?

Slivikkus, 2020-02-11:
[image]
C'est bien ce que je disais il est bien sur la partition OpenMPTCP mais ça fonctionne mal (lenteur probablement au niveau du VPN jsp)

dabuide, 2020-02-11:
fait un "netstat -tlpn"

Slivikkus, 2020-02-11:
L'accès au SSH est toute fois refusé

Slivikkus, 2020-02-11:
[image]

dabuide, 2020-02-11:
Je n'ai pas le détail en tête (je suis au boulot) mais visiblement shadowsocks n'est pas lancé !
lance un "netstat -anpe | grep 65101" pour vérifier

Slivikkus, 2020-02-11:
yep jte dit ça dans quelques minutes

Slivikkus, 2020-02-11:
Comment que je peux faire le `|` vu que j'ai accès qu'au KVM

dabuide, 2020-02-11:
Alt Gr + 6 ne fonctionne pas ?

Slivikkus, 2020-02-11:
Le clavier est en QWERTY sur le KVM

dabuide, 2020-02-11:
Shift + *

Slivikkus, 2020-02-11:
Le KVM est tellement bugguer que j'arrive pas à voir le résultat

Slivikkus, 2020-02-11:
ça met aucun résultat

dabuide, 2020-02-11:
voilà le problème, le mieux que je puisse te proposer c'est que ce soir en rentrant je te donne les commandes pour arranger la situation, voir le faire à distance, mais ici j'ai à rien vers l'extérieur impossible de voir les bonnes configurations serveurs ;)

Slivikkus, 2020-02-11:
D'accord

dabuide, 2020-02-11:
en attendant essaye tout de même "/etc/init.d/shadowsocks start" et si tu n'as aucun message relance le "netstat -tlpn" pour vérifier qu'il y a le port 65101 en écoute

Slivikkus, 2020-02-11:
[image]
[image]

dabuide, 2020-02-11:
On voit bien le démarrage du shadowsocks mais rien n'y fait il n'écoute rien voir se coupe tout seul, rdv d'ici 1h on va régler ce petit problème

dabuide, 2020-02-11:
cela semble différent ton ss-server n'est pas comme le mien il devrait être en écoute 65101

Ysurac, 2020-02-11:
shadowsocks-manager est maintenant utilisé (donc shadowsocks ne doit pas être lancé via autre chose), il se lance via ```systemctl start shadowsocks-libev-manager@manager```. La config de shadowsocks est dans ```/etc/shadowsocks-libev/manager.json``` et pour savoir si il y a un soucis, c'est dans ```/var/log/daemon.log```.

Normalement tu ne devrais pas avoir de soucis pour te connecter en SSH sur le port 65222 qui est bien en écoute.

dabuide, 2020-02-11:
Merci pour les précisions d'ailleurs "ss-server[463]: remote recv: Connection reset by peer" ce message n'est pas sensé être bon il me semble ?

Ysurac, 2020-02-11:
Quand il arrive de temps en temps ce n'est pas forcément un soucis, en permanence si.

Slivikkus, 2020-02-11:
Le port SSH est bien en écoute mais le serveur refuse la connexion, je me contente du KVM pour l'instant

Slivikkus, 2020-02-11:
[image]
C'est tout ce que je peux avoir

Ysurac, 2020-02-11:
Ce n'est pas le bon ss-server. Il faut arrêter celui la.
Essaye de redémarrer sinon.

Pour SSH il n'y a pas de raison que tu n'as pas accès, c'est bien un VPS normal Kimsufi ou OVH sous Debian 10 ?

Slivikkus, 2020-02-11:
Redémarrer le VPS (si c'est ça je l'ai déjà fait plein de fois)
VPS OVH sous Debian 10

dabuide, 2020-02-11:
tu utilises le mot de passe fourni dans le mail OVH ?

Slivikkus, 2020-02-11:
Oui je le change pas

Ysurac, 2020-02-11:
Tu as quoi exactement quand tu essayes de te connecter ? Le seul changement que je fais pour SSH c'est le port.

Slivikkus, 2020-02-11:
[image]

dabuide, 2020-02-11:
Il y a pas un souci de firewall avec debian ou selinux ?
tape "ufw status" et "getenforce" et donne moi le résultat

Slivikkus, 2020-02-11:
Y'a pas UFW ni gentenforce

dabuide, 2020-02-11:
et un "cat /etc/ssh/sshd_config"

Slivikkus, 2020-02-11:
Y'a le port de modifier, le root autorisé à se connecté

Slivikkus, 2020-02-11:
J'ai déjà regarder tout à l'heure

dabuide, 2020-02-11:
Tu as déjà essayé de te connecter avec un client SSH ? genre Putty

Slivikkus, 2020-02-11:
[image]

Slivikkus, 2020-02-11:
[image]

dabuide, 2020-02-11:
fait donc un "iptables -L | grep 65222"

Slivikkus, 2020-02-11:
[image]

Ysurac, 2020-02-11:
Le soucis ne pourrait pas être un firewall sur ton poste ou ton réseau ?
(sinon j'essaye une réinstallation sur un VPS OVH pour voir)

Slivikkus, 2020-02-11:
J'ai essayer sur un autre réseau que le mien mais ça fait la même chose

Ysurac, 2020-02-11:
Je viens de réinstaller un VPS OVH sous Debian 10 avec le script VPS sur https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS et après redémarrage j'ai pu me connecter via SSH par clef port 65222 sans soucis.

Slivikkus, 2020-02-11:
Moi pour le SSH c'est pas un problème, mais juste le problème de ShadowSocks qui est un peu énervant :/

Ysurac, 2020-02-11:
oui mais ça commence mal et tu as la moitié des services non lancés. Tu devrais peut être essayer une réinstallation. La dernière s'est terminée correctement ?

Slivikkus, 2020-02-11:
Exactement pareil ! :(

Slivikkus, 2020-02-11:
J'ai déjà réinstaller le VPS parce-que ça faisais la même erreur

dabuide, 2020-02-11:
je veux bien le faire avec toi si tu as moyen de partager en temps réel l'écran

dabuide, 2020-02-11:
Parce que ta situation est vraiment louche tout de même

Slivikkus, 2020-02-11:
J'ai teamviewer éventuellement

Ysurac, 2020-02-11:
Voila ce que tu dois avoir comme programme réseau lancés:
```
root@vps407669:~# netstat -laputen
Connexions Internet actives (seulement serveurs)
Proto Recv-Q Send-Q Adresse locale          Adresse distante        Etat        Utilisatr  Inode      PID/Program name    
tcp        0      0 0.0.0.0:65222           0.0.0.0:*               LISTEN      0          6642       501/sshd            
tcp        0      0 0.0.0.0:65001           0.0.0.0:*               LISTEN      0          6089       473/glorytun-tcp    
tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      0          5964       470/ss-server       
tcp        0      0 0.0.0.0:65011           0.0.0.0:*               LISTEN      0          5340       430/dsvpn           
tcp        0      0 0.0.0.0:65301           0.0.0.0:*               LISTEN      0          6657       488/openvpn         
tcp        0      0 0.0.0.0:65500           0.0.0.0:*               LISTEN      0          7777       457/python3         
tcp6       0      0 :::65222                :::*                    LISTEN      0          6653       501/sshd            
tcp6       0      0 :::65101                :::*                    LISTEN      0          5962       470/ss-server       
tcp6       0      0 :::65400                :::*                    LISTEN      0          4837       422/iperf3          
udp        0      0 0.0.0.0:67              0.0.0.0:*                           102        7278       284/systemd-network 
udp        0      0 0.0.0.0:67              0.0.0.0:*                           102        7250       284/systemd-network 
udp        0      0 0.0.0.0:67              0.0.0.0:*                           102        7222       284/systemd-network 
udp        0      0 0.0.0.0:68              0.0.0.0:*                           0          4280       351/dhclient        
udp        0      0 0.0.0.0:68              0.0.0.0:*                           0          4120       306/dhclient        
udp        0      0 127.0.0.1:323           0.0.0.0:*                           0          5572       455/chronyd         
udp        0      0 0.0.0.0:65001           0.0.0.0:*                           0          6182       481/glorytun        
udp        0      0 0.0.0.0:65101           0.0.0.0:*                           0          5968       470/ss-server       
udp        0      0 127.0.0.1:8839          0.0.0.0:*                           0          5970       462/ss-manager      
udp        0      0 0.0.0.0:65201           0.0.0.0:*                           108        6557       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65202           0.0.0.0:*                           108        6558       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65203           0.0.0.0:*                           108        6562       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65204           0.0.0.0:*                           108        6563       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65205           0.0.0.0:*                           108        6577       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65206           0.0.0.0:*                           108        6559       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65207           0.0.0.0:*                           108        6560       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65208           0.0.0.0:*                           108        6561       448/mlvpn: mlvpn0 ! 
udp        0      0 0.0.0.0:65301           0.0.0.0:*                           0          6751       496/openvpn         
udp6       0      0 ::1:323                 :::*                                0          5573       455/chronyd         
udp6       0      0 :::65101                :::*                                0          5966       470/ss-server
```

Ysurac, 2020-02-11:
Pour tester si le script d'admin fonctionne, tu peux aller sur https://<ipdetonvps>:65500/docs/

Slivikkus, 2020-02-11:
[image]

dabuide, 2020-02-11:
Effectivement il y a un souci avec la configuration le mieux est de reprendre à zéro

Slivikkus, 2020-02-11:
Zéro sur le VPS ? réinstaller du coup ?

dabuide, 2020-02-11:
Oui c'est mieux je peux aider par teamviewer si tu le souhaites

[137 more replies]

closed 2020-02-22 by Slivikkus